hzeller/rpi-rgb-led-matrix #807: add proper support for boards with  FM6126A chips

marcmerlin, 2019-05-02:
More panels are using those chips now, it would be nice if they were natively supported without a reset script like in https://github.com/hzeller/rpi-rgb-led-matrix/issues/746 and the issue of resetting boards on multiple strings with an active-3 board (as described in that bug).

I'm guessing that they an probably be reset properly as long as the reset code is sent to all 3 ports in parallel?

marcmerlin, 2019-05-02:
Ok, I did the above and it seems to work fine
Hopefully this can be included in the actual driver instead of being an out of band reset script

Here is one script that resets on 3 ports in parallel and another one that works on the adafruit board:
[resetmatrix_active3.py.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/3139328/resetmatrix_active3.py.txt)
[resetmatrix_adafruit.py.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/3139329/resetmatrix_adafruit.py.txt)

FelBraSil, 2019-06-20:
I'm using the init script but the panel just displays white lines, and they don't change, my previous panels work perfectly so wire problems are discarded

hyamanieu, 2019-07-05:
@marcmerlin thanks for your script on active3. It worked for initiating 4 pannels on a single chain, and 2 pannel in parallel on chains 1 & 2.
nb to all: you need to `sudo apt install python3-gpiozero` if you have lite raspian

marcmerlin, 2020-07-17:
and just to make sure things are too simple, FM6127 seems not fully compatible  with FM6126A despite a patch to support them. I opened a new bug to track that: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1103

marcmerlin, 2025-11-27:
this was fixed a long time ago

closed 2025-11-27 by marcmerlin