hzeller/rpi-rgb-led-matrix #808: Problem with latest P8 module of qiangli

hemanthkd, 2019-05-02:
I have done multiple iterations by changing the multiplexing number. But I was unable to get the desired image in any of the multiplexings. The latest version of Outdoor Qiangli P8 is having 40 LEDs in column & 20 LEDs in a row. Here is the command I have used :
Multiplexing=0,cols=40,rows=20, chain=1, parallel=2.Expected image & output images are as follows;
[image]
By the way I have used 2 panels.
[image]

jaygarcia, 2019-05-04:
@hemanthkd, sucks that you're having problems and thanks for posting a photo of your issue as well as some of the parameters you are using. What is the pin out of your panel and its refresh rate?  Simply saying that it's a `Qiangli P8` RGB Matrix isn't very helpful to anyone attempting to assist you :)

hemanthkd, 2019-05-05:
@jaygarcia, thanks for your advice. The panel is having 40 LEDs in horizontal,20 LEDs in vertical. The refreshing rate:>=1920Hz.Pin out of the panel is as follows:HUB75

1 | R1 | 2 | G1
-- | -- | -- | --
3 | B1 | 4 | N
5 | R2 | 6 | G2
7 | B2 | 8 | N
9 | A | 10 | B
11 | C | 12 | N
13 | S | 14 | L
15 | OE | 16 | CS

jaygarcia, 2019-05-07:
#778 Connecting this issue as it's a similar panel

hemanthkd, 2019-08-07:
the panel is having a different configuration. The IC used wad SM16207S. And none of the existing multiplexers is supporting for it. So I am working on developing the new multiplexer

caxepok, 2020-09-14:
> the panel is having a different configuration. The IC used wad SM16207S. And none of the existing multiplexers is supporting for it. So I am working on developing the new multiplexer

any news? :)

hemanthkd, 2020-11-10:
Actually no, but I made the sequence of the panel. Here I am attaching it. The panel consists of two halves with the same sequence
[Qiangli Q8.xlsx](https://github.com/hzeller/rpi-rgb-led-matrix/files/5519476/Qiangli.Q8.xlsx)

daveythacher, 2020-11-11:
This is an outdoor panel? 10 rows is kind of strange. Can you list the IC names on the back of the panel. Along with the number of ICs used. The sequencing how was this generated? Do you know the multiplex ratio? Have you verified if the A, B, C pins are used. I am not sure what is meant by S, L, and CS. However I am guessing S is CLK and L is LAT.

A strange configuration was found for outdoor panel. If it ends up being like that then you can follow the steps used for that one. It will require coding. See issue #1190.