hzeller/rpi-rgb-led-matrix #809: Raspberry Pi too busy. Cannot even connect through SSH.

pasveer, 2019-05-07:
Hello!

First a big thank you for this fantastic library! I am learning a lot of low level linux stuff! :-)

I have a "Raspberry Pi 3 Model B" running "Raspbian GNU/Linux 9 (stretch)" with 2 parallel chains of 6 modules. isolcpus=3 is set and works as expected. Only CPU 3 is 100% busy, the others are easy going. A few times per second, my application creates a bitmap and via a piped connection ledcat takes care of driving the matrix. This works fine. 

To make my commandline-life easier I launch my application+ledcat from a script (run.sh). Also this works fine.
`./MyLedMatrixScreen | sudo /home/pi/hzeller/ledcat --led-cols=64 --led-chain=6 --led-parallel=2 --led-pwm-bits=10 --led-pwm-lsb-nanoseconds=50 --led-slowdown-gpio=2`

The next step is to run my script at boot time, using /etc/rc.local :  my application starts and runs as expected. Nice!
My startup.sh script located in my home folder (called by /etc/rc.local) looks like this:
`cd /home/pi/hzeller
sleep 5
./run.sh 1>&- 2>&- &`

However... after powering up, once my application is up and running, the Pi seems so terribly busy that I cannot even establish a connection over SSH !  My ssh command is waiting endlessly for the Pi to respond. Then, the very moment my application exits (by setting some timer as a test) the ssh-process suddenly continues and prompts for my password... and I can login.

To allow me to connect over SSH before my application starts, I added a 5-second sleep in my script. This way I have access to my Pi before the show begins. But of course, I want to be able to login at any time.

I can reproduce the above with the original text_example when showing the current time like this:
`(while :; do TZ='Europe/Amsterdam' date +%n%c ; sleep 0.2 ; done) | sudo ./text-example --led-cols=64 --led-chain=6 --led-parallel=2 --led-pwm-bits=6 --led-pwm-lsb-nanoseconds=50 --led-slowdown-gpio=2 -f ./fonts/fonts/10x20.bdf -S 3 -x 76 -y -3 -C 50,0,0`

When I run this time.sh script at boot time, and I wait for about 20 seconds, I am again unable to establish a connection with my Pi.

Has somebody else seen this before? Why is my Pi so busy? It also seems that C system calls (like obtaining the IP address for example) is already too much to handle...

Thanks for your help!

jaygarcia, 2019-05-07:
Hi @pasveer i've run this using `isolcpus=3` and don't have the same problem on either a 3B or 3B+.  I am able to run  rsync jobs while my instance of the matrix is running.  I'm also using DietPi, which is way more optimized than Raspbian.  

Rather than run the program in the background using `&` , have you tried the `--led-daemon` switch?

angelogoncalve, 2019-05-07:
It is a good idea Jay, using the flag --led-daemon for acessing SO root in the RPi.

pasveer, 2019-05-09:
Thanks Jay and Angelo! I will do some tests (hopefully today) and look into DietPi as well and post the results.

jaygarcia, 2019-05-12:
Checking in to see if you've been able to test this. I'm jealously close to using that flag so am trying to learn from you before I take that step @pasveer  :D

hzeller, 2019-05-12:
Note that the Raspberry Pi has only one fairly slow memory bus, so if you saturate it with a realtime process, you'll get slow response with any other program e.g. with ssh etc. So even if the Pi is using only one core, it will saturate the whole system, as that thread updating the display runs at realtime priority and will starve out anything else.

With `--led-pwm-bits=6 --led-pwm-lsb-nanoseconds=50` on a 384 pixel wide display, this is definitely pushing it, as this means it will solidly push data without any chance of a slight break in-between.

It is made worse because we switch off a protective measure in the kernel which prevents it to get a breath. You can try to comment out the line that calls `DisableRealtimeThrottling()` in lib/gpio.cc: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/gpio.cc#L375
(this might result in super-slight regular flicker, but that is the price to pay). Recompile and then reboot before testing so that the setting is back to normal.

It looks like you are trying to push the limits as the refresh rate goes down with such long panel. I suggest trying the `--led-pwm-dither-bits` option for that and don't quite push down the lsb-nanoseconds quite as small.

pasveer, 2019-05-13:
Thank you Henner. All you write makes sense, especially the saturation part... poor Pi ;-)  In my example I even use `--led-pwm-bits=10` in order to have a good brightness control when reducing intensity down to 1%.

I will try the `DisableRealtimeThrottling()` suggestion and check if the flickering is acceptable. If 2 parallel chains of 6 modules is too much, I will see if I can use 3 parallel chains of 4 modules. 

Perhaps a silly question, but does the/each CPU has some 'near' memory that it can assess fast (and outside of the 'external' memory bus)? If so, would it be large enough to hold one (or two) screen buffers? 

@jaygarcia Perhaps my initial message was a bit unclear, but it was not my intention to suggest that the `isolcpus=3` flag is part of the problem. I just described my configuration.

pasveer, 2019-05-13:
I have done several tests today. To exclude anything silly introduced by myself I first downloaded & built the latest rpi-rgb-led-matrix-master code.

**SCRIPTS INVOLVED**

**/etc/rc.local** contains:
```
sleep 10
/home/pi/startup.sh
exit 0
```

**startup.sh** contains:
```
cd /home/pi/hzeller
sleep 5
./time.sh &
```

**time.sh** contains:
`(while :; do TZ='Europe/Amsterdam' date +%n%H:%M:%S ; sleep 0.2 ; done) | sudo ./text-example --led-cols=64 --led-chain=6 --led-parallel=2 --led-pwm-bits=6 --led-pwm-lsb-nanoseconds=50 --led-slowdown-gpio=2 -f ./fonts/fonts/6x10.bdf -S 3 -x 0 -y 0 -C 50,0,0` 

with text-example built with `DisableRealtimeThrottling() still included` in gpio.cc .

**INITIAL RESULT: confirmation the problem still exists**

At boot **rc.local** will wait 10 seconds and call **startup.sh**. **startup.sh** will again wait 5 seconds and then start **time.sh** in the background. I am able to SSH within the first 15~30 seconds, after that (when the date/time is shown for about 15 seconds on the screen) I cannot. The time is stable with acceptable flicker.

When I do have an SSH connection (connected during the startup delays), the busy-ness reveals itself as I even cannot kill text-example using `sudo kill nnn`. `sudo kill` simply hangs...

**TEST 1: use --led-daemon **
Adding  `--led-daemon` to the command line in **time.sh** has the effect that the date/time demo simply does not work anymore. Also my own application + ledcat won't start (or die immediately?) when passing `--led-daemon`. I don't understand why. Additionally note: even without `--led-daemon`, `top` tells me `text-example` and `ledcat` run as a daemon...?!

**TEST 2: comment DisableRealtimeThrottling() **
Commenting the DisableRealtimeThrottling() line + rebuild **helped** to get access by SSH again! Also killing text-example is no longer a problem. But the flicker is massive! I would call it a short blackout each ~1 second! See: [20190513_flicker_wIth_commented_DisableRealtimeThrottling.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/3172940/20190513_flicker_wIth_commented_DisableRealtimeThrottling.zip)

**TEST 3: reduce screen size **
Reducing the screen size to 50% (2 parallel chains of 3 modules == 192x64) did not help. Still no access via SSH. Cannot kill either. Also 128x64 does not give me anything.

**TEST 4: default --led-pwm-lsb-nanoseconds + use --led-pwm-dither-bits=2 **
I don't see any difference and it does not fix the issue.

**TEST 5: running the non-piped example demo **
All my tests involved launching 2 applications piped via stdout and stdin. `demo` is a stand-alone app... worth the try! I can launch demo with `--led-daemon` by the way.... but still no luck!

Until now the only serious change was commenting DisableRealtimeThrottling(), but the huge flicker is unacceptable. The saturation theory seems to be the most logic explanation. 

I am rather stuck now... perhaps I should start with a totally blank Raspbian image (and do some tests with DietPi as well).

Thanks for all your efforts... need to get my kid from school... I'll get back with more results later.

angelogoncalve, 2019-05-13:
Dear PasVeer. Try doing in the test 2:

Commenting the method DisableRealtimeThrottling() 
and changing the flag and reduce to the minimum:
--led-pwm-bits=1

pasveer, 2019-05-14:
> Dear PasVeer. Try doing in the test 2:
> 
> Commenting the method DisableRealtimeThrottling()
> and changing the flag and reduce to the minimum:
> --led-pwm-bits=1

Does not show any improvement.

But... there is news to share.... (next comment)

pasveer, 2019-05-14:
**With DietPi the problem has gone away!!** (of course new challenges in return)

Thanks @jaygarcia for the tip!!

Yesterday I downloaded the DietPi image for RaspberryPi and flashed it on a blank SD card. Almost immediately after the first boot, without any special tweaking (not even isolcpus), I copied my scripts + the demo binary.... and it worked! On the very same Pi hardware! I only swapped the SD card.

So what is the difference then?

My **trouble** Linux version:
`cat /etc/os-release`
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

`uname -a`
Linux raspberrypi 4.9.60-v7+ #1048 SMP Fri Nov 3 16:05:21 GMT 2017 armv7l GNU/Linux

My **DietPi** version:
`cat /etc/os-release`
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

`uname -a`
Linux DietPi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

The new problem is that many libs are missing on DietPi, like
    libpng16.so
    libharfbuzz.so
    libdouble-conversion.so
    libgthread-2.0.so
    libglib-2.0.so
and I don't know where to get them from... they seem so obvious.

Beersatron, 2019-05-14:
Have you tried Raspbian Stretch Lite? I use that for my scoreboard which is 2 parallel chains of 6 panels (32x64 per panel) and am able to SSH into the pi while it is running the matrix and also hosting an apache/php server - as my remote interface. It do not need any extra libs except for this hzeller recommends.

angelogoncalve, 2019-05-15:
Good choice Beersatron for the SO to put in the SD Card in the RPi - Raspbian Stretch Lite. And yes is not necessary extra libs except for the hzeller library to install in the pi CPU.

jaygarcia, 2019-05-15:
@pasveer, what are you using that requires those SO libs?

You could install them via something like `sudo apt-get install libpng16-16 -y`  (that's for libpng).  I don't recall this library requiring them.

jaygarcia, 2019-05-15:
Also, i do suggest adding `isolcpus=3` to your cmdline.txt file though ;)

pasveer, 2019-05-15:
Thanks for thinking with me @jaygarcia !

My application is Qt based and has those dependencies. I follow these [instructions](https://wiki.qt.io/RaspberryPi2EGLFS) to setup my system for the Raspberry Pi. Works perfectly!

Without success I tried to do the same with the DietPi image last Tuesday (at Step 5 to be exact). And if I recall correctly, I have tried to create a system based on Raspbian Stretch Lite as @Beersatron suggested months ago, but since that failed too at the time, I put it aside as I **thought** I had a nice & stable system already... sadly this is not the case.

PS: yes certainly, the `isolcpus=3` is a must, but without those SO files I have nothing ;)

jaygarcia, 2019-05-16:
I am curious. Is this issue (not being able to ssh in) happening when you run the Qt app also?

davemaster, 2019-05-16:
what rpi do you have? rpi 2, rpi3?

pasveer, 2019-05-17:
@jaygarcia
It happens regardless I run a Qt application or one of the "pure c/c++" examples included with ledmatrix.

@davemaster
I am using the Raspberry Pi 3 model B.

jaygarcia, 2019-05-17:
@pasveer this is indeed very strange and am wondering if you have a hardware problem. I've run examples on a 3B (not +) and don't have this problem what so ever. 

So just to be clear (sorry for making you repeat yourself), on a vanilla DietPi instance, you run one of these examples and you can't SSH in? 

Is this through Cat5 or wifi?

pasveer, 2019-05-17:
Today I downloaded burned the latest **Raspbian Stretch Lite** onto a fresh SD card.

a) After that I enabled SSH and modified `config.txt` to disable audio. I also appended `isolcpus=3` to `/boot/cmdline.txt` .

b) I reboot the Pi, remote login with SSH. Nice!

c) Then I copied the cross-compiled lib + examples + fonts to the Pi.

d) I created the scripts as I posted earlier (rc.local, startup.sh and time.sh).

e) Reboot... fingers crossed... booting... time.sh is started... 
... the ledmatrix shows the current time...
... but I CANNOT access SSH anymore !

@jaygarcia I always use CAT5 (no WiFi) and I can reproduce this on 3 Raspberry Pi 3B's I have.

jaygarcia, 2019-05-21:
hi @pasveer DietPi is reeeeaaaaally minimal.    As far as why the SSH issue is no longer present, i can't tell based on our interactions.  Raspbian uses OpenSSH (from what I can tell), and Diet Pi uses [DropBear](https://matt.ucc.asn.au/dropbear/dropbear.html).  I'm no expert in this arena.

As far as the missing libraries, you should be able to install them via:

`sudo apt-get install libpng16-16 libharfbuzz-dev libdouble-conversion-dev libglib2.0-0`

Though, it might be better to install the entire [Qt 5](https://wiki.qt.io/Qt5_dependencies)  -- assuming your app is using Qt5 specifically.  Else, you'll need to install [Qt4](https://askubuntu.com/a/866413) dependencies.

pasveer, 2019-05-22:
Hi @jaygarcia I strongly believe the SSH-issue I have is just a symptom, because:

a) I often have trouble `killing` the demo (or time) application. The demo keeps on running, but I have no longer access (hangs) to the remote shell. Also when I do a `sudo halt`... the demo example won't die and my Pi keeps running (I am not disconnected either!). The signals just don't seem to arrive...

b) My Qt application can show the IP address on the led matrix. I call [QNetworkInterface::allAddresses()](https://doc.qt.io/qt-5/qnetworkinterface.html#allAddresses) for this. When I call this function at startup of my application: **no problem**. But when my application is up and running (0.1 second later), my application **totally crashes** when I call `QNetworkInterface::allAddresses()` !  It seems some system calls are too much to handle.

In my previous post I wrote the easy steps to reproduce the trouble using Raspbian Stretch Lite. I hope somebody is able to do this test and share his/her results. There must be a simple answer... but I cannot see it. Or it's just because I am living 3 meters below sea-level...

PS: I will try to make Qt work on DietPi. Thanks for the hints!

hzeller, 2019-07-07:
I am suspicious that you mention Qt and the raspberry Pi. Usually Qt is used for graphical users interfaces and when using the Pi with the LED matrix, it is strongly not suggested to use any graphical UI as the Pi stimply can't handle it. Or are you just using Qt for non-UI stuff ? That could work.

pasveer, 2019-07-07:
I can understand mentioning "Qt" can triggers suspicion... but using Qt is **not** the cause here.

In order to exclude Qt as a suspect, the first thing I did was running the original `text-example` and the `demo` example: it all results in the very same behavior. My post of May 17 shows the few steps needed to reproduce the issue.

**THE GOOD NEWS IS** DietPi does not have these issues!!

The only trouble I **had** with DietPi is that my Qt application wouldn't start. `ldd` shows all dependencies are there, but when a Qt-GUI application starts many libraries are loaded dynamically at runtime and DietPi does not come with these libs (which totally makes sense by the way).

The reason I am using Qt is that I have old Qt-based code that I am using, and these drag in the QtGui stuff...  

@jaygarcia  Since about a week ago I managed to find & install all missing libs on my **DietPi** and I can finally run my Qt based application! `strace` helped me to find the missing libs.

**DietPi =** 
No SSH issues, no issues when doing system calls, no issues when trying to kill processes! And last but not least: instead of just a lousy 4 hours max my application runs for 8 days non-stop without any issues! Fast boot time, stable, .... etc etc I LOVE IT

And you are correct, doing GUI-stuff with the LED matrix will definitely cause flicker. But as long as the GUI does not need to update anything, the Pi has no trouble at all running Qt with a graphical interface.

oleliabo, 2020-05-02:
On a Pi3 / rasbian 4.19.66-v7+ I see similar problems as described above. Some time after starting in daemon mode and using --led-pwm-lsb-nanoseconds 50 and --led-pwm-dither-bits=1 sudo, ifconfig etc stops working. Also it does not accept new ssh connections. Established ssh terminal sessions still works. On my system the problem seems to be related to dmesg logs like this:

```
[ 3315.671977] INFO: task kworker/u8:2:151 blocked for more than 120 seconds.
[ 3315.671993]       Tainted: G         C        4.19.66-v7+ #1253
[ 3315.671996] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 3315.672002] kworker/u8:2    D    0   151      2 0x00000000
[ 3315.672039] Workqueue: netns cleanup_net
[ 3315.672073] [<80830934>] (__schedule) from [<80830fa4>] (schedule+0x50/0xa8)
[ 3315.672088] [<80830fa4>] (schedule) from [<80834f8c>] (schedule_timeout+0x200/0x428)
[ 3315.672110] [<80834f8c>] (schedule_timeout) from [<80831c14>] (wait_for_common+0xd4/0x1b0)
[ 3315.672121] [<80831c14>] (wait_for_common) from [<80831d10>] (wait_for_completion+0x20/0x24)
[ 3315.672135] [<80831d10>] (wait_for_completion) from [<8013a7e4>] (__flush_work+0x140/0x1e0)
[ 3315.672149] [<8013a7e4>] (__flush_work) from [<8013a8a0>] (flush_work+0x1c/0x20)
[ 3315.672161] [<8013a8a0>] (flush_work) from [<807038e4>] (rollback_registered_many+0x1d0/0x57c)
[ 3315.672179] [<807038e4>] (rollback_registered_many) from [<80703e48>] (unregister_netdevice_many+0x28/0xa4)
[ 3315.672187] [<80703e48>] (unregister_netdevice_many) from [<8070401c>] (default_device_exit_batch+0x158/0x17c)
[ 3315.672208] [<8070401c>] (default_device_exit_batch) from [<806f6c8c>] (ops_exit_list+0x6c/0x74)
[ 3315.672231] [<806f6c8c>] (ops_exit_list) from [<806f834c>] (cleanup_net+0x1c4/0x314)
[ 3315.672247] [<806f834c>] (cleanup_net) from [<8013bf0c>] (process_one_work+0x170/0x458)
[ 3315.672259] [<8013bf0c>] (process_one_work) from [<8013c250>] (worker_thread+0x5c/0x5a4)
[ 3315.672270] [<8013c250>] (worker_thread) from [<8014253c>] (kthread+0x138/0x168)
[ 3315.672287] [<8014253c>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[ 3315.672292] Exception stack(0xb9211fb0 to 0xb9211ff8)
[ 3315.672298] 1fa0:                                     00000000 00000000 00000000 00000000
[ 3315.672324] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 3315.672333] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
```

but I have not yet found a plausible explanation or a remedy. I run with `isolcpus=3` and sound disabled.

hzeller, 2020-05-02:
led-pwm-lsd-nanoseconds=50 is fairly small, and you will likely run solidly with 100% cpu without any return to the kernel. Unfortunately, we can't tell the kernel to not schedule anything on the core we'd like to reserve for us (or is there ? If someone knows, that would be great).

The problem likely occurs because we tell the kernel to not take any 'breathing' time to do its kernel tasks here:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/gpio.cc#L425

If you comment out this line, does it keep working for you @oleliabo ? What if you set the `-1` to `999000` ?

I suggest to not run with such low PWM nanoseconds; you want to tweak your values that you maybe get to 99% CPU, but not 100%.

oleliabo, 2020-05-02:
Thanks! I see indeed 100% cpu and `vcgencmd get_throttled` shows cpu throtteling. So I will tweak the parameters to get below 100%. 

Replacing -1 with 999000 in gpio.cc/425 seems to solve the problem.
https://access.redhat.com/solutions/1604133 seems to indicate that -1 should be avoided and https://www.kernel.org/doc/Documentation/scheduler/sched-rt-group.txt warns about using too low values. So this is a explanation of what I'm seeing I think.

hzeller, 2020-05-02:
Ok, made the default now `999000` instead of `-1`.

oleliabo, 2020-05-03:
My overnight test is still running with 100% CPU load to try to provoke the issue, so this seems to fix it on my system, thanks!

hzeller, 2020-05-03:
Sounds good. So I guess we can close this bug a few days short of its one year birthday :)

closed 2020-05-03 by hzeller