hzeller/rpi-rgb-led-matrix #811: Rectangle on Image

dbourdeveloper, 2019-05-12:
Using a Raspberry Pi 3 model v1.2 and Adafruit's Bonnet

Not sure if its my wiring, but every time I run any demo, I get a large white rectangle as shown in the image.

[image]

jaygarcia, 2019-05-12:
@dbourdeveloper Please post your matrix type, refresh rate, wiring and how you've connected it to the rpi, etc.. More input :D 

[image]

dbourdeveloper, 2019-05-12:
> @dbourdeveloper Please post your matrix type, refresh rate, wiring and how you've connected it to the rpi, etc.. More input :D
> 
> [image]

@jaygarcia , updated the OP. Thanks for getting back!

hzeller, 2019-05-12:
This just looks lie your LED matrix is broken.

dbourdeveloper, 2019-05-12:
> first 16 bit of the shift-register that supplies the bottom part is still working. So I suspect that there is a buffer or connection broken on the LED matrix that connects shift-registers chips between left and right with each other. Possibly just the clock line as it is harder to mess up 6 RGB lines in parallel. Inspect the leads on chips in the given quadrant closely and see if there is maybe a broken solder joint that might need a touch-up.

Thanks @hzeller , I was afraid that would be a case. I'll see what I can find but thank you for your amazing contribution with this library!

closed 2019-07-07 by hzeller