hzeller/rpi-rgb-led-matrix #812: Rock 64 GPIO /de/mem mmap error: Operation not permitted

krishnasandeep10694, 2019-05-14:


hzeller, 2019-05-14:
This library is for the Raspberry Pi, it will not work with any other SBC.

krishnasandeep10694, 2019-05-14:
Ok thanks for quick reply.

7Ds7, 2019-05-27:
Kind of getting the same results on a Rpi3 B+ running arch

```
sudo examples-api-use/demo -D0                                                                                                                                                                                       
mmap error: : Operation not permitted
RPi1: MMapping from base 0x20000000, offset 0x200000
Must run as root to be able to access /dev/mem
Prepend 'sudo' to the command
```

Current arch version is ```Linux alarm 5.1.5-1-ARCH #1 SMP Sat May 25 13:23:49 MDT 2019 aarch64 GNU/Linux``` i know i had this running on a RPi a few years ago although the arch version might have been 32 instead of 64bits, wonder if that might be the problem ?

hzeller, 2019-05-27:
Have a look, I think they changed things on recent kernels so that it is not possible to direcly mount /dev/mem. You have to figure out how to circumvent that.

7Ds7, 2019-05-27:
Thank you for the hing @hzeller yep seems that the only way to disable ```CONFIG_STRICT_DEVMEM``` would be building the kernel from source.

Changed to Raspbian Stretch Lite distribution and it works great

closed 2019-05-29 by krishnasandeep10694

quantenProjects, 2021-02-28:
I found a workaround for kernels with [`CONFIG_STRICT_DEVMEM` for my alpine linux on a raspberry pi](https://git.alpinelinux.org/aports/tree/main/linux-rpi/config-changes-rpi.armhf#n17):

The kernel parameter `iomem=relaxed` disables the address checking. Adding it to the `cmdline.txt` worked perfectly.

https://stackoverflow.com/a/49138256

btw: Alpine with its diskless mode is perfectly for display/presentation purposes, since the sd card is mounted read only and power losses doesn't corrupt the filesystem

phongans, 2024-01-18:
> Thank you for the hing @hzeller yep seems that the only way to disable `CONFIG_STRICT_DEVMEM` would be building the kernel from source.
> 
> Changed to Raspbian Stretch Lite distribution and it works great

How to rebuild new OS?