hzeller/rpi-rgb-led-matrix #814: Flickering

KeshavaShreenidhi, 2019-05-16:
Hello 

I have 7 tiles of led panels connected together in a row. I want to use these 7 tile in a manner that, for some time all the 7 tile should be used for displaying and after some time only 2 tiles (from right) should be used for displaying. So I integrated two 74HC245N chip to separate these 2 tiles. When i am using only 2 tiles it is working perfectly, but when i am using all the 7 tiles i get flickering in last 5 tiles. And I also tried to use --led-slowdown-gpio=4 flickers were slightly reduced. Is there any way to completely stop it. Please help me.

I attached a picture in which you can actually see the flicker.

[image]

angelogoncalve, 2019-05-16:
Try to use these flags:

--led-slowdown-gpio=1

--led-slowdown-gpio=4

jaygarcia, 2019-05-16:
Also, I wonder if the flicker can be dealt with by adding a capacitor.

KeshavaShreenidhi, 2019-05-17:
Hey angelogoncalve
Thank you for the reply. I have tried these but I can't see any changes.

KeshavaShreenidhi, 2019-05-17:
Hey jaygarcia 
Thank you for the suggestion. Where should I have add capacitor?

jaygarcia, 2019-05-17:
@KeshavaShreenidhi , please see: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power

hzeller, 2019-07-07:
Yes, this looks like a power issue. You _might_ be able to modify it with a different `--led-slowdown-gpio` value.

closed 2019-07-07 by hzeller