hzeller/rpi-rgb-led-matrix #815: Send text via Command

Thor421, 2019-05-16:
Hello and thank you very much hzeller for your work!

As in this subject, https://github.com/hzeller/rpi-rgb-led-matrix/issues/678, I want to send a text via a simple command line.

Unfortunately, echo "TEST" | sudo ./text-example ... does not work
Maybe a change has occurred in the bookstore this past year?

I specify that the matrix works very well by doing
sudo ./text-example -f ../fonts/8x13.bdf
Hello

Thank you for your help

Thor421, 2019-05-16:
small precision: this command line works but "TEST" remains 0.1 seconds in the matrix.

Thor421, 2019-05-16:
New precision : 
I use the fonction while from this topic : 
`(while :; do cat Anzeige.txt ; sleep 0.2 ; done) | sudo ./text-example -f ../fonts/7x13.bdf --led-rows 16 --led-gpio-mapping=adafruit-hat --led-cols 64 -y 1 -C 255,0,0
`
But, I can not clear the matrix. With CTRL+C, I have some points that stay on.
Is there a solution to clear all the matrice please ?

Thank you :)

angelogoncalve, 2019-05-16:
Use matrix.Clear()

[examples.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/3188020/examples.zip)

angelogoncalve, 2019-05-16:
from rgbmatrix import RGBMatrix, RGBMatrixOptions, graphics
import time

# Configuration for the matrix
options = RGBMatrixOptions()
options.rows = 16
options.cols = 64
options.chain_length = 1
options.parallel = 1
options.hardware_mapping = 'adafruit-hat'

matrix = RGBMatrix(options = options)

offscreen_canvas = matrix.CreateFrameCanvas()
font = graphics.Font()
font.LoadFont("../../../fonts/7x13.bdf")
textColor = graphics.Color(255, 0, 0)
my_text = "TEST"

len = graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, my_text)
time.sleep(5)
offscreen_canvas.Clear()

Thor421, 2019-05-16:
Thank you angelogoncalve :)
I adapted the code for my material but there is no display.

Can you help me please ?

```
from rgbmatrix import RGBMatrix, RGBMatrixOptions, graphics
import time
Configuration for the matrix

options = RGBMatrixOptions()
options.rows = 16
options.cols = 64
options.chain_length = 1
options.parallel = 1
options.hardware_mapping = 'adafruit-hat'

matrix = RGBMatrix(options = options)

offscreen_canvas = matrix.CreateFrameCanvas()
font = graphics.Font()
font.LoadFont("../../../fonts/7x13.bdf")
textColor = graphics.Color(255, 0, 0)
my_text = "TEST"

len = graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, my_text)
time.sleep(5)
offscreen_canvas.Clear()
```

Thor421, 2019-05-17:
I'm progressing slowly and I have a new problem to solve.

Can I use anything other than a CTRL-C to stop the display of text? I looked at SIGINT but I do not see how to adapt this function to python code.

Can you help me please ?
Thank you !

Thor421, 2019-05-17:
Maybe his information will be useful.

1- I run: sudo ./text.py (it shows me a static text)
2- I wish to interrupt 1
3- To launch: sudo ./text2.py (it shows me a static text)
I do not use a keyboard, only a box with pushbuttons mounted on an arduino card that communicates in bluetooth with the raspberry (a python file manages the reception of bluetooth data)

How to interrupt 1? I hope it's more understandable :)

hyamanieu, 2019-05-18:
@Thor421 you need to import signal and os then do 
`os.killpg(os.getpgid(pid_of_your_prog), signal.SIGTERM)`
you need to start your command from another python script to do that and have its PID saved.
You can use `p = subprocess.Popen(['python3','text.py], cwd=yourpath, preexec_fn=os.setsid)` to do that, the pid is then `p.pid`.

angelogoncalve, 2019-05-18:
you can use killing threads to clear the panel (stop RPi sending data to the display):

https://eli.thegreenplace.net/2011/12/27/python-threads-communication-and-stopping

https://www.geeksforgeeks.org/python-different-ways-to-kill-a-thread/

Thor421, 2019-05-18:
Thanks for your help hyamanieu and angelogoncalve :)
I just coded the lines to kill the process from the PID

`os.killpg(os.getpgid(pid), signal.SIGINT)`

Now, I think I have to modify the samplebase.py file and especially this part to add this line, but how ?

```
        try:
            # Start loop
            print("Press CTRL-C to stop sample")
            self.run()
        except KeyboardInterrupt:
            print("Exiting\n")
sys.exit(0)
```

Thank you so much !

Thor421, 2019-05-19:
My problem :
Python does not execute any command while a text is displayed on the matrix. The commands made (when I press the push button) while displaying the text (including os.killpg) are done after I have done CTRL + C.

Of course, if I manually call the python file to send the SIGINT signal, everything works.

Could you give me an additional indication to solve this problem please?

```
ser = serial.Serial('/dev/rfcomm0', 9600) // Connect to the bluetooth
while True:
	result = ser.read()
	print(result)	           	

         if result == 'f' : 		
    		
		os.system("sudo python ./30.py")		
		
```

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828