hzeller/rpi-rgb-led-matrix #819: Can I have one chain composed of 2 rows of LED Displays ?

GeorgeFlorian, 2019-05-22:
Hello !

I am currently using a Raspberry Pi Zero W V1.1. I am using it just to test things out until I get my hands on a Raspberry Pi 2 or 3.

I am using 64x32 1/8 SMD P5 2727 LED Panels.

I am trying to have 60 LED Panels arranged as follows:
```
_P6_P5_P4_P3_P2_P1_Chain1_RaspBerryPi1
P11_P10_P9_P9_P8_P7_
_P6_P5_P4_P3_P2_P1_Chain2_RaspBerryPi1
_P11_P10_P9_P9_P8_P7
_P6_P5_P4_P3_P2_P1_Chain3_RaspBerryPi1
_P11_P10_P9_P9_P8_P7
_P6_P5_P4_P3_P2_P1_Chain1_RaspBerryPi2
_P11_P10_P9_P9_P8_P7
_P6_P5_P4_P3_P2_P1_Chain2_RaspBerryPi2
_P11_P10_P9_P9_P8_P7
```

Basically 10 rows with 6 panels each.
**My question is how can I make 1 chain to be on 2 rows and how can I tell one row to be 6 displays in length ?**

Until now I've linked 4 displays in 1 chain on 1 row. How can I tell this chain to be on 2 rows and to be 2 panels in length ?

One last question: is it possible to have the above arrangement run by a single Raspberry Pi ?

This would mean 2 chains with 30 panels each. 
```
_P6_P5_P4_P3_P2_P1_Chain1_RaspBerryPi1
_P11_P10_P9_P9_P8_P7_
_P17_P16_P15_P14_P13_P12_
_P23_P22_P21_P20_P19_P18_
P29_P28_P27_P26_P25_P24_
_P6_P5_P4_P3_P2_P1_Chain2_RaspBerryPi1
_P11_P10_P9_P9_P8_P7_
_P17_P16_P15_P14_P13_P12_
_P23_P22_P21_P20_P19_P18_
P29_P28_P27_P26_P25_P24_
```

Beersatron, 2019-05-22:
I think you will need a Raspberry Pi 3 for this. I created a scoreboard that has 12 64x32 panels in two parallel chains.

My cmd is:

sudo /home/pi/rpi-rgb-led-matrix/examples-api-use/scoreboard -f /home/pi/rpi-rgb-led-matrix/fonts/10x20.bdf --led-pixel-mapper="U-mapper" --led-rows=32 --led-cols=64 --led-chain=6 --led-parallel=2 --led-show-refresh --led-slowdown-gpio=2 --led-pwm-dither-bits=2

This is the part that makes 6 chained panels into "2 rows of 3 panels":
--led-pixel-mapper="U-mapper" --led-chain=6

GeorgeFlorian, 2019-05-24:
> I think you will need a Raspberry Pi 3 for this. I created a scoreboard that has 12 64x32 panels in two parallel chains.
>
> My cmd is:
>
> sudo /home/pi/rpi-rgb-led-matrix/examples-api-use/scoreboard -f [...]

Thank you for your answer.

For 6 panels on one row I will use the following, correct ?
```
--led-pixel-mapper="U-mapper" --led-rows=32 --led-cols=64 --led-chain=12 --led-parallel=3
```
Basically, with this I can split one chain just once, resulting in 3 chains with 12 panels each that will be split just once using the U-map. This means 36 panels.

I need 10 rows with 6 panels each, so 60 panels.
This would mean that I will need a second Raspberry Pi 3 with another 2 chains with 12 panels each, for a total of 10 rows.

But can I compose the final image using two separate Raspberries ?

Also, is there any way I can use just one Raspberry ?
```
_P6_P5_P4_P3_P2_P1_Chain1_RaspBerryPi1
_P11_P10_P9_P9_P8_P7_
_P17_P16_P15_P14_P13_P12_
_P23_P22_P21_P20_P19_P18_
P29_P28_P27_P26_P25_P24_
_P6_P5_P4_P3_P2_P1_Chain2_RaspBerryPi1
_P11_P10_P9_P9_P8_P7_
_P17_P16_P15_P14_P13_P12_
_P23_P22_P21_P20_P19_P18_
P29_P28_P27_P26_P25_P24_
```

```
[<][<][<][<][<][<]-Chain1_RaspBerryPi1
[>][>][>][>][>][>]
[<][<][<][<][<][<]
[>][>][>][>][>][>]
[<][<][<][<][<][<]
[>][>][>][>][>][>]-Chain2_RaspBerryPi1
[<][<][<][<][<][<]
[>][>][>][>][>][>]
[<][<][<][<][<][<]
[>][>][>][>][>][>]

```
I would need to split 1 chain 4 times to make it into 5 rows of panels.
How can I translate that into a cmd ?

`--led-rows=32 --led-cols=64 --led-chain=30 --led-parallel=2 --led-pixel-mapper="S-shape"`

Or maybe 3 chains:

```
_P6_P5_P4_P3_P2_P1_Chain1_RaspBerryPi1
_P11_P10_P9_P9_P8_P7_
_P17_P16_P15_P14_P13_P12_
P23_P22_P21_P20_P19_P18_
_P6_P5_P4_P3_P2_P1_Chain2_RaspBerryPi1
_P11_P10_P9_P9_P8_P7_
_P17_P16_P15_P14_P13_P12_
P23_P22_P21_P20_P19_P18_
_P6_P5_P4_P3_P2_P1_Chain3_RaspBerryPi1
_P11_P10_P9_P9_P8_P7_
```

How would the cmd look like ? I would be Chain 1 in an S-mapping, Chain 2 also in an S-mapping and Chain 3 in U-mapping
```
[<][<][<][<][<][<]-Chain1_RaspberryPi1
[>][>][>][>][>][>]
[<][<][<][<][<][<]
[>][>][>][>][>][>]
[<][<][<][<][<][<]-Chain2_RaspberryPi2
[>][>][>][>][>][>]
[<][<][<][<][<][<]
[>][>][>][>][>][>]
[<][<][<][<][<][<]-Chain3_RaspberryPi2
[>][>][>][>][>][>]
```

In this case I would need a way of choosing the mapping for each chain separately.

angelogoncalve, 2019-05-24:
see this link: 

https://github.com/t413/rpi-rgb-led-matrix/tree/s-mapper

flag:

--led-pixel-mapper="S-mapper"

shanerooni, 2020-03-02:
> I think you will need a Raspberry Pi 3 for this. I created a scoreboard that has 12 64x32 panels in two parallel chains.
>
> My cmd is:
>
> sudo /home/pi/rpi-rgb-led-matrix/examples-api-use/scoreboard -f [...]

@Beersatron , I'm trying to also create a 12 panel 64x32 display.  would you be able to walk me through some pointers?  if possible, you can email me using the email on my profile.  thanks!

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828