hzeller/rpi-rgb-led-matrix #820: Funky new P10 1/4 Scan (so the product description says)

grene78, 2019-05-22:
A little confused by all the various panel issues, but I think I have a new one not covered by any other issues.  What is the current method for handling these oddball displays?  Is there a tutorial on starting from scratch?  If not, I'll write one as I go.  My setup works perfectly with p6 indoor and p10 indoor but this p10 outdoor is a whole different beast.  Any pointers are appreciated.

2 x 16x32 P10 1/4 Scan Panels

sudo ./demo -D 3 --led-rows=16 --led-cols=32 --led-chain=2 --led-multiplexing=0
[image]

sudo ./demo -D 3 --led-rows=16 --led-cols=32 --led-chain=2 --led-multiplexing=1
[image]

sudo ./demo -D 3 --led-rows=16 --led-cols=32 --led-chain=2 --led-multiplexing=5
[image]

Configuring in the controller that we currently use for these panels (but trying to move away from dedicated controllers and use the system built on a pi.  I couldn't get the whole thing because of the nature of the crummy software, but this gives an idea.
[image]

Product Link: https://www.aliexpress.com/item/1634189398.html?spm=a2g0s.9042311.0.0.65b34c4dqJWAh1

grene78, 2019-05-23:
Never mind, these panels are already supported.

For anyone else that comes across this, fresh eyes as I started to play around with the multiplex mapper classes led me to the discovery of --led-row-addr-type which I admit I somehow totally missed in my earlier endeavors.

sudo ./demo -D 3 --led-rows=16 --led-cols=32 --led-chain=2 **--led-row-addr-type=2** --led-multiplexing=5

[image]

angelogoncalve, 2019-05-23:
Yes you can see in the readme to put the flag:

--led-row-addr-type=2

for 16x32 1/4 Scan Rate panel

closed 2019-06-03 by grene78