hzeller/rpi-rgb-led-matrix #821: Panel Orientation Help

oliverdarvall, 2019-05-23:
A few questions with regards to panel orientation:

1. 
I have 8 32x16 panels that are connected into 2 parallel chains of 4 each, thus ending up with 128x32 display.

I would really prefer to install the panels in Portrait orientation and using a big font then display a single digit per panel. What I would like to achieve is the following:

[image]

Is this possible to achieve using the "Rotate" options ? 

2.
If I have 2 parallel chains of boards installed with 4x (32x16) board per chain, to give me an effective resolution of 128x32. Is it possible to then set the library to "handle" it as a 64x64 display, thus "breaking up" the long chain into shorter "chains" ?

oliverdarvall, 2019-05-23:
Oh my ! Is the U-mapper the answer to my question 2 ?

angelogoncalve, 2019-05-23:
Yes maybe can be the flag:

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828