hzeller/rpi-rgb-led-matrix #822: Qiangli q6 32x32 display not working

logaeshKTBS, 2019-05-25:
I have a Qiangli q6 32x32 display. When i connect it with raspberry pi & Active adapter and run a sample sketch in python it shows nothing. I tried with different parameter by tuning it in command line and double checked the connections and power but no luck. But display works fine with D10 LED board controller.
Here is the q6 display's specification
[image]

angelogoncalve, 2019-05-25:
Try these flags:

--led-rows=32 --led-cols=32 --led-multiplexing=1

logaeshKTBS, 2019-05-29:
I have already tried all the flags. but no luck. I suspect the driver for this issue. Because when I tried this LED panel with D10 controller Initially it won't lit the display. After updating the D10 Controller with sm16237d frimware, the display function properly. What may the possible issue?

Sabia123, 2019-11-28:
Hi, logaeshKTBS

Did you get a D10 controller based display working with the repository code? I am looking at buying such a display and I need some code so that I can generate data in a custom app (or even a browser tab) and send to the display over LAN. I's appreciate any input you may have :-) Thanks!

logaeshKTBS, 2019-11-30:
Hi @Sabia123 ,
Sorry for delayed reply. I have worked with 3 different type of p6 displays from different manufacturers. All the three works fine with D10 controller (you may have to update D10's firmware given by display manufacturer). But for Raspberry pi out of 3 two works fine and I have no luck in one display which i have posted in this issue. Even after tuning various parameters as mentioned by @angelogoncalve i can't able to lit the display.

Sabia123, 2019-11-30:
Hi,

Thanks for the reply! I actually want to control a D10 based display using a custom PC app. It’s for a sports timer. The D10 connects to the PC over LAN

Du you know if the code on here can be used for that? Thanks again!

logaeshKTBS, 2019-11-30:
Yes. You can control a D10 based display with a raspberry pi (PC) using the code provided here.

Sabia123, 2019-12-03:
Hi again. I'm sorry to keep on about it but I need to make sure: I want to use the display with a D10 controller and a PC, connected via LAN. I don't want any raspberry pi whatsoever. Will the code on here work for that, or is the code on here meant to **replace** the D10 (or similar) controller card with a Raspberry Pi?

logaeshKTBS, 2019-12-04:
The code here is to replace the D10 controller card with a raspberry pi. You can't able to use this code in your PC.

punyanagari, 2020-02-18:
Did you get this working

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin