hzeller/rpi-rgb-led-matrix #823: Sigh, new panels: ABC with fm26126A

marcmerlin, 2019-05-25:
This is starting to get annoying...

[image]
[image]

jake653, 2019-06-03:
Previous discussions of the quirky  fm6126A"

https://github.com/hzeller/rpi-rgb-led-matrix/issues/746

There is some code and comments regarding that chip, in this driver
https://github.com/2dom/PxMatrix/blob/master/PxMatrix.h

One guy suggests that only init code must be added to solve his FM6126A problem

http://ledpixelart.com/pixelv2board/ declares the FM6126A is not compatible with their setup.

One user post:
"I just ported this file so that it works on my rPi, ran it, and then plugged my Teensy with SmartMatrix code (unmodified), and things just worked.
So this confirms that the only problem is indeed the init necessary with those FM6126A chips"

https://community.pixelmatix.com/t/smartmatrix-doesnt-support-fm6126a-driver-chips/421

marcmerlin, 2019-06-03:
"One guy suggests that only init code must be added " => that was me :)

Sorry that my message was unclear, the problem is not FM6126A which is solved with the out of band init code (hopefully will get integrated in the lib proper soon), it's the ABC addressing.
@hzeller  can you confirm that ABC addressing is not currently supported by your lib (or any other lib I've seen actually)?  I've only ever seen AB or ABCD or ABCDE

mcpgza, 2019-08-31:
> Sorry that my message was unclear, the problem is not FM6126A which is solved with the out of band init code (hopefully will get integrated in the lib proper soon), it's the ABC addressing.

I have ABC addressing panel using ICN2037, and I was not able to make it work.

hzeller, 2019-09-03:
If this is an FM61A but only has three address lines, would this probably be working with (the new flag) `--led-panel-type=fm6126` and one of the --led-multiplexing options (because outdoor panels do different multiplexing).

mcpgza, 2019-09-05:
My 64x128 panel is not fm6126, but ICN2037, and it has only ABC lines:

[image]

and this is how the demo with -D 3 looks like:
[image]

It is written on the panel: P2-128X64-32S-V4.0, LED1515, 2018/06/16

(I have other panels with HUB75E with ICN2037, and those works fine)

hzeller, 2019-09-06:
ABC sounds like a different led multiplexing. Did you try through the available `--led-multuplexing` options?

mcpgza, 2019-09-09:
Yes, I tried them all, none of them worked. 

This is how demo 0 looks like:

sudo ./demo  --led-cols=128 --led-rows=64 --led-slowdown-gpio=5   -D 0

[image]

hzeller, 2019-09-11:
ABC could also means some different row address type; there are a few implemented, but they don't do three address lines. It might be worthwhile doing some reverse engineering, then implement this.

It looks a bit like the AB row address type but maybe split into top and bottom half ? So I would try with `--led-row-addr-type=1` and see if that at least gives some partial usable output. Then we can go from there.

themultiplexer, 2019-09-13:
I also have one of these ABC-panels and have the same issue as @5onix, @marcmerlin and @mcpgza. 
I'd love to see this awesome library work with my panel. I would like to help but I don't know where to start 😅
Maybe I can help after I heard Computergraphics this semester.

hzeller, 2019-09-13:
Good starting point is to get hold of a datasheet

gustavoalara, 2019-09-19:
Hi, 

I have the same case as @mcpgza. I received two days ago two 128x64 led-matrix with the same ABC addresses and same information written on the panel (P2-128X64-32S-V4.0, LED1515, 2018/06/16). I asked the chinese store from Aliexpress where I bought the panel for some schematics or something useful but they said that their have no information about it

gustavoalara, 2019-09-19:
Hi again,

According to the adafruit site (https://learn.adafruit.com/32x16-32x32-rgb-led-matrix/new-wiring), the ABC addresses is the pin arrangement used by 32x16 panels
[image]

I don't know if this could be useful to someone. Maybe this panel needs a custom internal mapper...

bk313, 2019-09-19:
I also have this 128x64 panel with the ICN2037 and the ABC lines. No luck getting it to work, I get the same image as @mcpgza posted above. Would like to help any way I can.

rowanG077, 2019-09-19:
I have this panel as well. It seems a completely different addressing scheme then before.

mcpgza, 2019-09-20:
It seems it requres a new RowAddressSetter, and I would like to wrtire it, but unfortunately I dont understand how it works, dear @hzeller, can you provide a short descrption how it works, what it does. Thanks.

rowanG077, 2019-09-20:
I have looked at the code for the AB address setter. The way it seems to work is that `A` is a clock
signal and `B` is the data. You clock on the `A` signal and when you have reached the `addressableRows - rowToSelect` clock you set `B` to low for that clock cycle. As an example
if you have a panel with 8 addressable rows you and you want to select the third row you would first
clock the `A` line for 5(8 - 3) cycles while keeping `B` high and then you would clock the `A` line for a single cycle and have `B` low. And the clock `A` for the remaining two cycles while keeping `B` high again. Below is an illustration:

[image]

Here is the relevant [code](https://github.com/hzeller/rpi-rgb-led-matrix/blob/7160a8ade3dcecabe25ce0de73232b365f7587be/lib/framebuffer.cc#L127).
```
  virtual void SetRowAddress(GPIO *io, int row) {
    if (row == last_row_) return;
    for (int activate = 0; activate < double_rows_; ++activate) {
      io->ClearBits(clock_);
      if (activate == double_rows_ - 1 - row) {
        io->ClearBits(data_);
      } else {
        io->SetBits(data_);
      }
      io->SetBits(clock_);
    }
    io->ClearBits(clock_);
    io->SetBits(clock_);
    last_row_ = row;
  }
```

What I don't understand is  that extra clock low and clock high after the loop. Anyway this is just an analysis for the AB line setter and I don't have such a panel. The ABC addressing may be similar to this or completely different.

gustavoalara, 2019-09-20:
According to Adafruit for the 32x16 Led matrix,
 
A is the lowest bit of row address
B is the second lowest bit of row address
C is the highest bit of row address
LE is Latch of row
CLK is Clock
EN is  Enable
R1 is Red data 1
R2 is Red data 2
G1 is Green data 1
G2 is Green data 2
B1 is Blue data 1
B2 is Blue data 2
GND is Grounding (and 4 GND must be connected)

I think ours panels is the same pinout but with a 1:32 scan multiplexing... Anyway, I've contacted the manufacturer of my led matrix (Coreman.cc). Let's see if I'm lucky and they answer me with the information I've requested

rowanG077, 2019-09-20:
The addressing of the 32x16 Adafruit panels are the same as the usual A,B,C,D addressing but it's just using the A,B and C lines directly since it just needs to address 8 rows. This addressing on the panels here is completely different.

gustavoalara, 2019-09-20:
Hi,

I found this forum with some information that can be useful https://community.pixelmatix.com/t/led-module-p2-128-64-1-32-driver/414

rowanG077, 2019-09-20:
That could be it. I hadn't found that yet even though I searched that forum. Maybe I can take a look this weekend if this will work.

mcpgza, 2019-09-23:
I made a huge progress in refactor. the panel does not contain inverters, so it use "positive logic", and it seems tha panel has AB type addressing, but B is not used, C used instead B.

sudo ./demo --led-cols=128 --led-rows=64 --led-slowdown-gpio=5 --led-row-addr-type=1 -D 0
[image]

This is the relevant code: 
`class ShiftRegisterRowAddressSetter : public RowAddressSetter {
public:
  ShiftRegisterRowAddressSetter(int double_rows, const HardwareMapping &h)
    : double_rows_(double_rows),
      row_mask_(h.a | h.c),
      clock_(h.a),
      data_(h.c),
      last_row_(-1) {
  }
  virtual gpio_bits_t need_bits() const { return row_mask_; }

  virtual void SetRowAddress(GPIO *io, int row) {
    if (row == last_row_) return;
    for (int activate = 0; activate < double_rows_; ++activate) {
      io->SetBits(clock_);
      if (activate == double_rows_ - 1 - row) {
        io->SetBits(data_);
      } else {
        io->ClearBits(data_);
      }
      io->ClearBits(clock_);
    }
    io->SetBits(clock_);
    io->ClearBits(clock_);
    last_row_ = row;
  }

private:
  const int double_rows_;
  const gpio_bits_t row_mask_;
  const gpio_bits_t clock_;
  const gpio_bits_t data_;
  int last_row_;
};
`
The only problem is a missing line at the end of one part.

bk313, 2019-09-23:
Great work @mcpgza !

hzeller, 2019-09-23:
Thanks @mcpgza ! I have added your code to the master branch now and it can be selected with `--led-row-addr-type=3` so that it is easy to try for everyone.

This then allows others who also have the panel to help figure out the remaining issue with the missing line.

gustavoalara, 2019-09-25:
Hi guys,

I'm trying the new option for our 128x64 ABC modules. Apart of the black line, I see two things:

1. some LEDs with incorrect colours near the right and left laterals
2. If I try to connect 2 128x64 modules in chain, the modules don't show anything and the application freezes

I'll upload some images of these things

gustavoalara, 2019-09-25:
Forget the second issue... I incremented the nanoseconds from 300 to 450 and now the chain works fine. Only the of issues with some LEDs:

[image]

5onix, 2019-09-25:
For your 1st problem, have you tried the demo #4  - Pulsing color or #11 - Brightness pulse generator ?
Those demos made me realize that my panel had a few bad subpixels on the edges.
The edges are kinda fragile. I had to remove a few leds which weren't working anymore after some accidental bumps they were falling apart.

gustavoalara, 2019-09-25:
Hi @5onix I've tried with demos and images (as you can see in my previous message) and the result is the bad subpixels on the edges. My second matrix module is failing and the low quarter of the module has no light (I thing something has broken)

Following a picture of the two modules working together, (the left one is the broken module). I have 3 LEDs broken too (in the right corner of the right module)
[image]

5onix, 2019-09-25:
Too bad that one of your module has an issue. But overall the result looks very promising.
Here's 2 photos of mine which show broken and malfunctioning pixels.
[image]
[image]

marcmerlin, 2019-09-25:
speaking of P2 panels and pixels that fall off, have you found some kind of glue spray that you can layer on top of them to glue them all in place so that they don't get lost? (and without affecting display quality and/or the space between panels if you put them adjacent to one another)

gustavoalara, 2019-09-25:
Good point @marcmerlin, this panels look very fragile and is very easy to lost pixels. (I found 2 of the pixels, but I don't know how to glue them to the panel again)
In the other hand, I'm arguing with the seller because I think the panel arrived broken

marcmerlin, 2019-09-25:
@gustavoalara you need both. I also had a panel that was not protected correctly in shipping and arrived with some detached pixels, but trust me when I tell you that you will lose more unless you encase this in some way, or never touch/move it, so you might as well practice with the damaged panel :)

5onix, 2019-09-25:
Epoxy resin maybe ?

gustavoalara, 2019-09-26:
Then, you thing the LEDs with incorrect colours are damaged too? Well, maybe if I'm lucky the seller will send me another panel and I can practice with the damaged panel as @marcmerlin said...

5onix, 2019-09-26:
I know it's a little off topic but here's a little demo of the panel playing a video with VLC & sound enabled.
[https://gofile.io/?c=F2Mp86](url)

mcpgza, 2019-09-26:
Ok, here is the fully functional SetRowAddress (no missing line), dear @hzeller please update the code

```
 virtual void SetRowAddress(GPIO *io, int row) {

    for (int activate = 0; activate < double_rows_; ++activate) {
      io->ClearBits(clock_);
      if (activate == double_rows_ - 1 - row) {
        io->SetBits(data_);
      } else {
        io->ClearBits(data_);
      }
      io->SetBits(clock_);
    }
    io->SetBits(clock_);
    io->ClearBits(clock_);
    last_row_ = row;
  }

```

I dont know why is it works, but works.

5onix, 2019-09-26:
Awesome progress made here, thank you @mcpgza !

gustavoalara, 2019-09-27:
Hi!

The change proposed by @mcpgza works like a charm! My only problem now is the dead and malfunctioning pixels
[image]

[image]

marcmerlin, 2019-09-27:
For dead pixels, in the future, I was told to consider this
https://www.amazon.com/gp/product/B0009GXMJG
not sure how well it works

embedded-creations, 2019-09-30:
@marcmerlin If you need the LED to be functional, I think you'd need to hot air rework the connections, or replace the LED:

https://twitter.com/GregDavill/status/1172850535403114496

marcmerlin, 2019-09-30:
@embedded-creations nice. I wasn't the one who lost pixels on those panels (but I lost some on another panel, that were already detached when the panel arrived in the mail), but while I'm impressed by the work done to fix them, that's a bit beyond my equipment/skill level for the moment.
I'll focus on trying to find the right glue spray to put on top of the entire panels when I receive them so that hopefully I don't lose pixels to start with.
My immediate problem with my wearable panels is that because they are flexible, even the slight flex they receive eventually messes solder joints of the chips in the back, making me lose entire columns or rows depending on which pad gets disconnected (not in a visible way). Thankfully it's usually only one color component, so I don't lose the entire pixel, but it's still annoying.
This is out of topic for this bug, so I opened a new discussion here https://community.pixelmatix.com/t/rgb-panels-and-wear-issues-losing-pixels-and-losing-solder-joints-in-the-back-on-flexible-panels/530

hzeller, 2019-10-07:
Now it is :)
Thanks @mcpgza !

marcmerlin, 2019-12-23:
@mcpgza @hzeller I've only been able to try my ABC panels (really AC from what you said) now.
It works great with --led-row-addr-type=3 --led-panel-type=FM6126A
BTW, @hzeller  I can confirm --led-panel-type=FM6126A works with the active parallel board, both my panels were initialized perfectly.

whiteshavedchocolate, 2020-02-10:
Guys, I am pretty new to all of this. However, I have the datasheet for the ICN2037 in English. Would that be of interest/help to anybody?

It looks like the main feature it adds over a standard LED shift register/column driver is that it has "dual latch for higher refresh rate". Maybe that can confirm how @mcpgza got the ABC LED panel to light up properly.

marcmerlin, 2020-02-14:
@mcpgza @bk313 @gustavoalara @5onix can I ask you to have a look at https://github.com/hzeller/rpi-rgb-led-matrix/issues/985 (I'm somehow not allowed to @ you in that issue).
Long story short, ABC works fine, but it's slower than ABCDE, which it shouldn't be.

marcmerlin, 2020-02-14:
@whiteshavedchocolate this bug is about ABCDE vs ABC. Using the wrong led-row-addr-type still lights up fine, but displays lines in the wrong places.
ICN2037 vs FM6126A is a shift register issue, and not related to  led-row-addr-type
What is the exactly problem you have and you're trying to fix?

whiteshavedchocolate, 2020-02-14:
I don't currently have a problem. I am learning and interested in hardware
development around these Chinese LED panels. I am willing to help where I
can, but my skills are in hardware and very limited in software/firmware.

marcmerlin, 2020-02-14:
@whiteshavedchocolate ok, then there is no problem at the moment, ICN2037 seems to work fine as is (unless I missed something)

tdaede, 2020-02-14:
FWIW the current code implements 33 clocks into the shift register (an extra clock at the end), but I found that 32 clocks is correct for my FPGA reimplementation. I'm wondering if the last clock is fast enough that it's getting missed. (the last clock actually looks like a leftover from the old version with inverted clock polarity).

marcmerlin, 2020-02-23:
@mcpgza  @bk313  @gustavoalara  @5onix can I ask you where you bought your panels from, and what you paid for them?
My current supplier is offering them at $50 per panel (shipped from china), and unfortunately he says he can only sell ABC, as ABCDE doesn't seem to be available anymore. 
I'm a bit bummed about having to use ABC panels with a lower refresh rate, but if that's all what is available today, and $50/panel is about the right price, so be it, I'll go ahead and get those
(I need 8, so making sure before I plop down $400 on this)

mcpgza, 2020-02-23:
@marcmerlin I bought theese: https://www.aliexpress.com/item/32998082589.html
ABC panels, not perfect panels, some of them seem used, some have missing leds on the edge, but generally i'm happy with them for this price.  
But why do you think that ABC panels are slower then ABCDE?

marcmerlin, 2020-02-23:
Thanks @mcpgza . I'll probably pay a bit more for pristine panels, I don't want missing pixels for my setup :)
https://github.com/hzeller/rpi-rgb-led-matrix/issues/985 explains why ABC is slower than ABCDE. In a nutshell, ABC panels are ABCDE panels with 2 address lines missing, so it takes longer to send the 5 bits of addresses over 3 bits.
you can benchmark this yourself with
~/rpi-rgb-led-matrix/examples-api-use/demo --led-rows=64 --led-cols=128 --led-chain=1 --led-show-refresh --led-parallel=3   --led-pwm-dither-bits=1 --led-pwm-bits=7 --led-panel-type=FM6126A  --led-row-addr-type=0 -D0  and then use led-row-addr-type=3
You can do the speed test even without the right panels and look at the Hz value you get.

arash1221, 2020-04-24:
Hello @hzeller , First of thanks for your awesome code, please anyone can help me with my problem!!!
My problem is when I'm add another panel to the output of the first panel the screen shows me different output but alone with just 1 led panel everything is okay! and also I using the same led panel!! panel IC is ------> FM6126A_ 64*128
I tried these command flags also not work
 --led-chain=2  && --led-chain=1 
I also set col to 256 and also dosen't work!!!,And also I'm not using any hat, direct to raspberry pi GPIO! Everything is ok when its just only 1 panel no more chain!!!!!

Please Somebody Help!
Appreciate!

arash1221, 2020-04-24:
First Image Only 1 led Panel

[image]

Second Image two chain panel !!!

[image]

mcpgza, 2020-04-24:
> First Image Only 1 led Panel

> 
> Second Image two chain panel !!!
> 

Yeah, tipical, you need a level shifter, you need a hat.

arash1221, 2020-04-24:
No way to use hat??

arash1221, 2020-04-24:
@mcpgza thanks for the answer, can you please guide me how to control any pixel of led panel,
very simple code, I just want to control each pixel on my own, I tried a lot with nodejs but no result I get!
thanks for your help.

hzeller, 2020-04-24:
@arash1221 Please don't hijack an existing issue with an unrelated problem; open a new issue.

To your problem: try to gpio slowdown your output (`--led-slowdown-gpio`)  depending on the internal wiring you're increasing the load on the gpio pins, which will make them switch  slower. 

Also, you _really_ want a level shifter, as not only will be the digital voltage correct (and right now you're just essentially playing jeopardy if it might work), but also, you can't really drive long lines with the GPIO pins directly.

arash1221, 2020-04-25:
Thanks @hzeller  I recall it.

xsrf, 2020-05-06:
> Thanks @mcpgza . I'll probably pay a bit more for pristine panels, I don't want missing pixels for my setup :)
> #985 explains why ABC is slower than ABCDE. In a nutshell, ABC panels are ABCDE panels [...]

Hey, I've seen your discussion about ABC being slower than ABCDE. This is because the code is not using the Shift-Register that is connected to the row selection as it is meant to be used (I guess).

To select one out of N rows, you are shifting out N-1 zeroes and one one out to the display. Of course this needs more than N clock cycles which is way longer on 1/16 or even 1/32 panels than just setting 5 GPIOs at once.

However: This is actually not necessary if you scan your rows incrementally, which is usually what is done. In this case you actually do only have to shift out one bit for every cycle. You shift out 1 for row 0 and 0 otherwise.

Here is the code I contributed to PxMatrix few days ago:
https://github.com/2dom/PxMatrix/blob/178054b3021892cb9185c3a92051ba2b2afc8977/PxMatrix.h#L1012

If you don't require random row access, it boils down to:

```
      // Just shift the row mux by one for incremental access
      digitalWrite(_C_PIN, (value==0) ); // Shift out 1 for line 0, 0 otherwise
      digitalWrite(_A_PIN, HIGH); // Clock out this bit
      digitalWrite(_A_PIN, LOW);
```

Hope this may also help you :)

hzeller, 2020-05-06:
Nice, thanks xsrf

[24 more replies]

commit 21410d2b (2019-10-07): Fix ABC-type panels, thanks to @mcpgza

Fixes #823
closed 2025-05-07 by marcmerlin