hzeller/rpi-rgb-led-matrix #825: Compiling minimal-example.cc returns `undefined reference to`

GeorgeFlorian, 2019-05-28:
Hello !
I am trying to compile the `minimal-example.cc`, but I receive errors.

**Here are the commands written in Raspberry PI Terminal:**
```
mkdir 4panel
cd 4panel
git clone https://github.com/hzeller/rpi-rgb-led-matrix.git
sudo nano main.cpp
sudo g++ -Wall -Irpi-rgb-led-matrix/include  main.cpp -o main
```

**In main.cpp we find:**
```
#include "led-matrix.h"

#include <unistd.h>
#include <math.h>
#include <stdio.h>
#include <signal.h>

using rgb_matrix::GPIO;
using rgb_matrix::RGBMatrix;
using rgb_matrix::Canvas;

volatile bool interrupt_received = false;
static void InterruptHandler(int signo) {
  interrupt_received = true;
}

static void DrawOnCanvas(Canvas *canvas) {
  /*
   * Let's create a simple animation. We use the canvas to draw
   * pixels. We wait between each step to have a slower animation.
   */
  canvas->Fill(0, 0, 255);

  int center_x = canvas->width() / 2;
  int center_y = canvas->height() / 2;
  float radius_max = canvas->width() / 2;
  float angle_step = 1.0 / 360;
  for (float a = 0, r = 0; r < radius_max; a += angle_step, r += angle_step) {
    if (interrupt_received)
      return;
    float dot_x = cos(a * 2 * M_PI) * r;
    float dot_y = sin(a * 2 * M_PI) * r;
    canvas->SetPixel(center_x + dot_x, center_y + dot_y,
                     255, 0, 0);
    usleep(1 * 1000);  // wait a little to slow down things.
  }
}

int main(int argc, char *argv[]) {
  RGBMatrix::Options defaults;
  defaults.hardware_mapping = "regular";  // or e.g. "adafruit-hat"
  defaults.rows = 32;
  defaults.chain_length = 1;
  defaults.parallel = 1;
  defaults.show_refresh_rate = true;
  Canvas *canvas = rgb_matrix::CreateMatrixFromFlags(&argc, &argv, &defaults);
  if (canvas == NULL)
    return 1;

  // It is always good to set up a signal handler to cleanly exit when we
  // receive a CTRL-C for instance. The DrawOnCanvas() routine is looking
  // for that.
  signal(SIGTERM, InterruptHandler);
  signal(SIGINT, InterruptHandler);

  DrawOnCanvas(canvas);    // Using the canvas.

  // Animation finished. Shut down the RGB matrix.
  canvas->Clear();
  delete canvas;

  return 0;
}

```

**The errors are:** 
```
/tmp/cci8MGL5.o: In function `main':
main.cpp:(.text+0x264): undefined reference to `rgb_matrix::RGBMatrix::Options::Options()'
main.cpp:(.text+0x2a8): undefined reference to `rgb_matrix::CreateMatrixFromFlags(int*, char***, rgb_matrix::RGBMatrix::Options*, rgb_matrix::RuntimeOptions*, bool)'
collect2: error: ld returned 1 exit status

```

GeorgeFlorian, 2019-05-29:
I've also compiled the library by using `make`. This resulted in 2 files: `librgbmatrix.a` and `librgbmatrix.so.1`.

[image]

I've changed my commands but the `undefined reference to`errors still appear.

The commands that I've tried:
```
sudo g++ -Wall -Irpi-rgb-led-matrix/include -l./rpi-rgb-led-matrix/lib/librgbmatrix.so.1 main.cpp -o main
```
```
sudo g++ -Wall -Irpi-rgb-led-matrix/include -l/home/pi/4panel/rpi-rgb-led-matrix/lib/librgbmatrix.a main.cpp -o main
```
```
sudo g++ -Wall -Irpi-rgb-led-matrix/include -Lrpi-rgb-led-matrix/lib -lrgbmatrix -O3 -g -Wextra -Wno-unused-parameter main.cpp -o main
```
```
sudo g++ -Wall -Irpi-rgb-led-matrix/include -Lrpi-rgb-led-matrix/lib -lrgbmatrix -lrt -lm -lpthread -O3 -g -Wextra -Wno-unused-parameter main.cpp -o main
```

GeorgeFlorian, 2019-05-29:
I've made a makefile inside `4panel`.
```
CXXFLAGS=-Wall -O3 -g -Wextra -Wno-unused-parameter
OBJECTS=main.o
BINARIES=main

# Where our library resides. You mostly only need to change the
# RGB_LIB_DISTRIBUTION, this is where the library is checked out.
RGB_LIB_DISTRIBUTION=rpi-rgb-led-matrix
RGB_INCDIR=$(RGB_LIB_DISTRIBUTION)/include
RGB_LIBDIR=$(RGB_LIB_DISTRIBUTION)/lib
RGB_LIBRARY_NAME=rgbmatrix
RGB_LIBRARY=$(RGB_LIBDIR)/lib$(RGB_LIBRARY_NAME).a
LDFLAGS+=-L$(RGB_LIBDIR) -l$(RGB_LIBRARY_NAME) -lrt -lm -lpthread

all : $(BINARIES)

$(RGB_LIBRARY): FORCE
	$(MAKE) -C $(RGB_LIBDIR)

main: main.o $(RGB_LIBRARY)
	$(CXX) $(CXXFLAGS) main.o -o $@ $(LDFLAGS) $(MAGICK_LDFLAGS)

%.o : %.cc
	$(CXX) -I$(RGB_INCDIR) $(CXXFLAGS) -c -o $@ $<

main.o : main.cc
	$(CXX) -I$(RGB_INCDIR) $(CXXFLAGS) $(MAGICK_CXXFLAGS) -c -o $@ $<

clean:
	rm -f $(OBJECTS) $(BINARIES) $(OPTIONAL_OBJECTS) $(OPTIONAL_BINARIES)

FORCE:
.PHONY: FORCE
```
4panel is the directory that contains `pi-rgb-led-matrix` directory and `main.cc`

Then I've ran 
`sudo ./main --led-rows=32 --led-cols=64 --led-multiplexing=1 --led-chain=4 --led-no-hardware-pulse`

And the displays light up and they seem to work. But I have no idea how `minimal-example.cc` should look.

hzeller, 2019-05-29:
Just watch what the makefile executes:
 - it comipiles the main object function
 - it then links this with the library to make an executable;

```
$ cd examples-api-use/
$ make minimal-example 
g++ -I../include -Wall -O3 -g -Wextra -Wno-unused-parameter -c -o minimal-example.o minimal-example.cc
g++ minimal-example.o -o minimal-example -L../lib -lrgbmatrix -lrt -lm -lpthread
```

What you are trying to do is to directly compile the main.cc to an executable, but you are missing the libraries to link. So you need to provide the `-L` and `-l` options in that case

GeorgeFlorian, 2019-05-29:
> Just watch what the makefile executes:
>
> * it comipiles the main object function
> * it then links this with the library to make an executable;
>
> ```
> $ cd examples-api-use/
> $ make minimal-example
> g++ [...]

And wasn't this enough ?
```
sudo g++ -Wall -Irpi-rgb-led-matrix/include -Lrpi-rgb-led-matrix/lib -lrgbmatrix -lrt -lm -lpthread -O3 -g -Wextra -Wno-unused-parameter main.cpp -o main
```

closed 2019-05-30 by GeorgeFlorian

izy521, 2019-07-08:
You closed this, but I'm having the same issue and following what was listed on the examples page. What was your solution?