hzeller/rpi-rgb-led-matrix #827: 80*40 Size issue

krishnasandeep10694, 2019-05-29:
Hi @hzeller,
I am using an 80*40 panels with this library. When I connect 4 80*40 panels in u shape the size is determined correctly as you can see below

> pi@raspberrypi:~/rpi-rgb-led-matrix/utils $ sudo ./led-image-viewer --led-rows=40 --led-chain=4 --led-cols=80  --led-multiplexing=1 /home/pi/160_80.jpg  --led-pixel-mapper="U-mapper;Rotate:180"
> Size: 160x80. Hardware gpio mapping: regular
> Loading 1 files...
> Loading took 0.049s; now: Display.

But when i use 6 80*40 panels in u shape the size is determined as 224*80 instead of 240*80

> pi@raspberrypi:~/rpi-rgb-led-matrix/utils $ sudo ./led-image-viewer --led-rows=40 --led-chain=6 --led-cols=80  --led-multiplexing=4 /var/tmp/ad_images/2_E856amS.jpg  --led-pixel-mapper="U-mapper;Rotate:180"
> Size: 224x80. Hardware gpio mapping: regular
> Loading 1 files...
> Loading took 0.061s; now: Display.

Any help would be appreciated. Thanks

krishnasandeep10694, 2019-05-31:
Can someone help me in this issue please?

Beersatron, 2019-05-31:
What happens when you remove "Rotate:180" and just leave it as U-mapper?

krishnasandeep10694, 2019-06-01:
Even If I remove the "Rotate:180" the size is being determined as 224*80
```
pi@raspberrypi:~/rpi-rgb-led-matrix/utils $ sudo ./led-image-viewer --led-rows=40 --led-chain=6 --led-cols=80 --led-pixel-mapper="U-mapper" --led-multiplexing=1 /var/tmp/ad_images/2_E856amS.jpg
Size: 224x80. Hardware gpio mapping: regular
Loading 1 files...
Loading took 0.058s; now: Display.
```
But when I remove "U-mapper" and run it the size is being determined as 480*40 which is correct.
```
pi@raspberrypi:~/rpi-rgb-led-matrix/utils $ sudo ./led-image-viewer --led-rows=40 --led-chain=6 --led-cols=80 --led-pixel-mapper="Rotate:180" --led-multiplexing=1 /var/tmp/ad_images/2_E856amS.jpg
Size: 480x40. Hardware gpio mapping: regular
Loading 1 files...
Loading took 0.031s; now: Display.
```
I think there is some limit on pixels number if we use "U-mapper"

Thanks for Reply

krishnasandeep10694, 2019-06-01:
I found that in U mapper the matrix width must be a multiple of 32 as they are setting boundary at 32 px.
I made some tweaks to U mapper code and fixed it.

closed 2019-06-01 by krishnasandeep10694