hzeller/rpi-rgb-led-matrix #830: class "rgb_matrix::Canvas" has no member "SetBrightness"

GeorgeFlorian, 2019-05-30:
Hello !
As the title says I can't use `SetBrightness` function.

I am using `using namespace rgb_matrix;` and I've included `"led-matrix.h"` so it should be there. But it isn't.

This is the canvas:
`Canvas *canvas = rgb_matrix::CreateMatrixFromFlags(&argc, &argv, &defaults);`

This is the code:
```
#include "led-matrix.h"
#include "graphics.h"

#include <iostream>
#include <unistd.h>
#include <math.h>
#include <stdio.h>
#include <string.h>
#include <signal.h>

using namespace rgb_matrix;

volatile bool interrupt_received = false;
static void InterruptHandler(int signo) {
  interrupt_received = true;
}

const char *fontFile = "rpi-rgb-led-matrix/fonts/10x20.bdf";

static void TextLine(Canvas *canvas, int x, int y, const char* line) {
    canvas->Fill(225, 20, 34);
    rgb_matrix::Color mBlue(0, 0, 255);
    rgb_matrix::Color blackBg(0, 0, 0);
    rgb_matrix::Font font;
    if(!font.LoadFont(fontFile)) {
        std::cout<<"Couldn't load font file\n";
    }
    if (interrupt_received)
      return;
    rgb_matrix::DrawText(canvas, font, x, y, mBlue, &blackBg, line, 0);
    for (int i = strlen(line); i>0; i--) {
        usleep(100);
    }
}

int main(int argc, char *argv[]) {

    RGBMatrix::Options defaults;
    defaults.hardware_mapping = "regular";  // or e.g. "adafruit-hat"
    defaults.rows = 32;
    defaults.chain_length = 1;
    defaults.parallel = 1;
    defaults.show_refresh_rate = true;
    Canvas *canvas = rgb_matrix::CreateMatrixFromFlags(&argc, &argv, &defaults);
    if (canvas == NULL)
        return 1;
    // It is always good to set up a signal handler to cleanly exit when we
    // receive a CTRL-C for instance. The DrawOnCanvas() routine is looking
    // for that.
    signal(SIGTERM, InterruptHandler);
    signal(SIGINT, InterruptHandler);
    canvas->SetBrightness(40);
    TextLine(canvas,1,10,"Test 1");

    canvas->Clear();
    delete canvas;

    return 0;
}
```

jaygarcia, 2019-05-30:
Suggestion:

Add `defaults.brightness = 40`;
Remove `canvas->SetBrightness(40)`;

GeorgeFlorian, 2019-05-30:
Ok ! That worked !
But what I want is a setter ! I want to be able to change the brightness via input.

jaygarcia, 2019-05-30:
looks like you may be casting inappropriately:  
Try:
`RGBMatrix *matrix = rgb_matrix::CreateMatrixFromFlags(...)`

GeorgeFlorian, 2019-05-31:
> looks like you may be casting inappropriately:
> Try:
> `RGBMatrix *matrix = rgb_matrix::CreateMatrixFromFlags(...)`

Changing from 
```
Canvas *canvas = rgb_matrix::CreateMatrixFromFlags(&argc, &argv, &defaults);
```
to 
```
RGBMatrix *canvas = rgb_matrix::CreateMatrixFromFlags(&argc, &argv, &defaults);
```
made the `SetBrightness` function work.

But will `RGBMatrix` work the same as `Canvas` ?
I've seen that `RGBMatrix` inherits `Canvas` : `class RGBMatrix : public Canvas` and I assume that the above declarations work fairly similar, at least in my case.

jaygarcia, 2019-05-31:
Awesome! @GeorgeFlorian . Please close this issue =)

closed 2019-06-01 by GeorgeFlorian