hzeller/rpi-rgb-led-matrix #831: 120x60 flexible led matrix.

ostap-dm, 2019-06-02:
Hi, I'v just bought this matrix https://www.aliexpress.com/item/32961595666.html
and I couldn't get it worked. I'v use this command and get such picture. I tried any different types of cols and rows.
Can you help me with were to look for solving this problem. Thank you. 
[Result with ./led-image-viewer --led-rows=60 --led-cols=120 --led-gpio-mapping='adafruit-hat-pwm' --led-slowdown-gpio=3  jp.png](https://github.com/ostap-dm/img/blob/master/photo_2019-06-02_13-29-45.jpg)

[source image](https://github.com/ostap-dm/img/blob/master/jp.png)

[led pannel](https://github.com/ostap-dm/img/blob/master/photo_2019-06-02_13-29-48.jpg)
[led pannel](https://github.com/ostap-dm/img/blob/master/photo_2019-06-02_13-29-51.jpg)

davemaster, 2019-06-02:
Try other multiplexing

ostap-dm, 2019-06-02:
I tried every multiplexing. And it's not working.

davemaster, 2019-06-03:
have You panels arrangement? Size?

ostap-dm, 2019-06-03:
120x60 LEDs SMD P2. size is 240x120mm

ostap-dm, 2019-06-03:
https://www.aliexpress.com/item/32961595666.html 
here is full information

jake653, 2019-06-03:
The troubling spec is:  **Moudle Scan | 30scan**

(you may share issues with the recent 40x20 reference, both being non powers of 2 columns)

angelogoncalve, 2019-06-04:
Try this flags in the command:

./led-image-viewer --led-rows=60 --led-cols=120 --led-gpio-mapping='adafruit-hat-pwm' --led-slowdown-gpio=3 --led-multiplexing=0 jp.png

ostap-dm, 2019-06-04:
This is output result

ostap_dm@mail.ru
On Jun 4, 2019, 10:26 AM -0400, hzeller/rpi-rgb-led-matrix <reply@reply.github.com>, wrote:
>
> ./led-image-viewer --led-rows=60 --led-cols=120 --led-gpio-mapping='adafruit-hat-pwm' --led-slowdown-gpio=3 --led-multiplexing=0 jp.png

ostap-dm, 2019-06-08:
So, none of the advises works. 
Is there any other ways to solve this problem?

ostap-dm, 2019-06-13:
[image]
[image]
[image]

That's how sudo ./demo  -D0  --led-cols=120 --led-rows=60 looks on my panel

hzeller, 2019-07-07:
Looks like you're getting close and might need to do a little bit of position mapping.

ostap-dm, 2019-07-07:
I gave up on it and bought module for using such led matrix. It turned out that it has 1/30 scan

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin