hzeller/rpi-rgb-led-matrix #832: Issue with Qiangli 64x64 RGB panel

zaun, 2019-06-08:
I'm having some issues with my [panel](https://www.amazon.com/gp/product/B07LFJ73GQ) that I bought. As far as I can tell its this [specific](http://www.qiangliled.com/products-4.html) panel, though I posted a picture of the back of mine below. I've tried to run the demos it kinda works, but not really. and after a few moments it goes wonky. Here is a quick [video](https://www.youtube.com/watch?v=sb7GYp-yhoo) of demo 0.

I've made a little test program to light up LED `0,0` with only red.
```
while(1) {
  led_canvas_set_pixel(offscreen_canvas, 0, 0, 1000 & 0xff, 0, 0)
  offscreen_canvas = led_matrix_swap_on_vsyc(matrix, offscreen_canvas);
}
```

I'm running with `--led-rows=64 --led-cols=64` and I've tried `--led-rows=64 --led-cols=64 --led-row-addr-type=1` and `--led-rows=64 --led-cols=64 --led-row-addr-type=2`. If I try either row address type 1 or 2 the matrix remains blank.

With just the `0,0` led coded I get 4 LEDs lit. In addition after a few seconds I get a line of LEDs lit.

I'm using the [Adafruit HAT](https://learn.adafruit.com/adafruit-rgb-matrix-plus-real-time-clock-hat-for-raspberry-pi/overview). I've soldered the E-8 connection. I've tried E-16 but the panel remains blank.

If it matters, [this](https://www.amazon.com/gp/product/B07C58BLYY) is my power supply.

[image]
[image]
[image]

zaun, 2019-06-08:
I can't read it, but the Datasheet for the chip [SM5266P](https://datasheetspdf.com/pdf-file/1328896/Sunmoon/SM5266P/1) is online in CN if it helps.

zaun, 2019-07-16:
Any idea where to begin supporting these?

jaygarcia, 2019-08-01:
Hi @zaun, rather than starting with a custom demo, have you tried using the compiled `demo` program inside of `examples-api-use/` to test?

zaun, 2019-08-02:
@jaygarcia, yes, none of them work properly. That why I thought I would try just a really basic program.

jaygarcia, 2019-08-02:
Hi @zaun, i imagine this won't provide a ton of value, but holy crap -- those panels are SOOOO overpriced.  For context, I got 24 ea. 64x64 32s P3 HUB75 panels from china for about $300 shipped. :(  I've also ordered RGB Panels from amazon and they were not as advertised, and wildly expensive. I eventually hunted down a reliable vendor in China and can refer you to him if you'd like.  We've built a complex large screen using 20 panels.

I wonder if the panels themselves are just non-functional or perhaps the pins are not labeled correctly.  I have two panels, 192mm^2 labeled HUB75 and 160mm^2 labeled HUB75E. Each have different pin labeling and I do have them connected and they work perfectly in serial. 

One thing can do to test is to see if pin 8 is an E line by testing resistance with ground as and you can do the same for pin16.  It's likely that 8 is truly E. Trying to create some idea to help you solve =).

This is an image of the two panels I have connected. I'm trying to get the video demo uploaded, however my upstream bandwidth is crap atm.
[image]

zaun, 2019-08-03:
@jaygarcia - i just ordered some from eBay, but I’d be interested if your supplier as well.

jaygarcia, 2019-08-03:
The video of the above setup running in my office/lab.
https://www.youtube.com/watch?v=1aDurOOmkeo

I purchase from `Hoonshine` and have worked with Paul Zhang. He's shipped to our offices in the US and Romania for us.  Takes about 2 weeks, but worth it for a good supplier. 

https://www.alibaba.com/product-detail/SMD-full-color-3in1-rgb-led_60839777344.html?spm=a2700.7724838.2017115.14.2eec76delIFz9O

Depending on the order configuration, I've gotten prices from $18.75 to $22 / panel shipped.

[image]

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828