hzeller/rpi-rgb-led-matrix #833: Issue with Adafruit Hat and RPi3

pboehm90, 2019-06-12:
Hello!
I have an big issue with my LED setup. I have an RPI3, an Adafruit LED Hat and four HUB75 16x32 Panels. I have connected these four panels in a chain. I had compiled the led-matrix library with the adafruit-hat comiler flag, but i cant get it working. Because not even the demos working properly I wrote an simple program with just iterates the X and Y Axis with the setpixel method, i made an video that you can see whats going on. Does anybody have an idea please i tried so much...

I made a video of the problem and uploaded to Youtube
[Click here!](https://youtu.be/Z2OudXlwKqk)

Thank you very much

Beersatron, 2019-06-12:
What is the full cmd line that you are using when triggering the demos?

pboehm90, 2019-06-13:
I wrote an C# program which use the C# bindings of this library, In the Minute 1:15 in the video you can see the settings i passed to the library.

hzeller, 2019-07-07:
From the video output this looks like an unsupported panel. Do you have a datasheet or chipset for the panel ?

hzeller, 2019-07-17:
Can this be closed ?

pboehm90, 2019-07-21:
Sorry for the late answer. Unfortunately, I could not get a data sheet. But i finnally get it work :), i had to enable the **--led-multiplexing=3** option.

jaygarcia, 2019-08-02:
@pboehm90 please close this issue when you have a moment =)

closed 2019-08-02 by pboehm90