hzeller/rpi-rgb-led-matrix #834: Nothing outputs on Banana Pi M3

GeorgeFlorian, 2019-06-18:
Hello !

I've just moved my project from a Raspberry Pi Zero, where all worked perfectly to a Banana Pi M3, and nothing works.

Here are the settings I run in the code:
```
int main(int argc, char *argv[]) {

  RGBMatrix::Options defaults;
  defaults.hardware_mapping = "regular";  // or e.g. "adafruit-hat"
  defaults.multiplexing = 1;
  defaults.cols = 64;
  defaults.rows = 32;
  defaults.chain_length = 4;
  defaults.parallel = 1;
  defaults.pixel_mapper_config = "U-mapper";
  defaults.disable_hardware_pulsing = true;
  defaults.show_refresh_rate = false;
  RGBMatrix *canvas = rgb_matrix::CreateMatrixFromFlags(&argc, &argv, &defaults);
  if (canvas == NULL)
      return 1;
//...
}
```

I had to add `-std=c++11` to the makefile from whatever reason.
The wiring is the same.
The LED Panels stay black.

Does anybody have any idea ?

GeorgeFlorian, 2019-06-19:
I've changed the OS to Armbian Stretch:
```
Linux bananapim3 4.19.38-sunxi
PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=debian
```

This is `lsmod`:
```
pi@bananapim3:/boot$ lsmod
Module                  Size  Used by
evdev                  20480  1
axp20x_adc             16384  0
axp20x_battery         16384  0
axp20x_usb_power       16384  0
industrialio           49152  3 axp20x_usb_power,axp20x_battery,axp20x_adc
leds_axp20x            16384  0
zstd                   16384  8
sunxi_cir              16384  0
cpufreq_dt             16384  0
uio_pdrv_genirq        16384  0
uio                    16384  1 uio_pdrv_genirq
thermal_sys            57344  1 cpufreq_dt
zram                   24576  2
ip_tables              20480  0
x_tables               20480  1 ip_tables
uas                    20480  0
pwrseq_simple          16384  0
pinctrl_axp209         16384  4
realtek                16384  1
```

I am fairly positive that the Banana Pi M3 is so bad that it doesn't have a Kernel Sound Module.

hzeller, 2019-07-07:
This is expected. The Banana Pi is entirely different hardware, it will not work with this library unless you change the gpio.cc mapping. I am happy to accept pull requests if you have provided a mapping.

Till then, this library will only work with the Raspberry Pi.

closed 2019-07-07 by hzeller

GeorgeFlorian, 2019-07-07:
> This is expected. The Banana Pi is entirely different hardware, it will not work with this library unless you change the gpio.cc mapping. I am happy to accept pull requests if you have provided a mapping.
> 
> Till then, this library will only work with the Raspberry Pi.

Ok. So what you are saying is that the pins on the Banana Pi do NOT do the same thing as the Raspberry Pi ?
The problem is that any way I look at gpio.cc I understand nothing. Do you have any starting points, helping advice... anything to start me off ?

depili, 2019-07-07:
On Sun, 7 Jul 2019, GeorgeFlorian wrote:

> Ok. So what you are saying is that the pins on the Banana Pi do NOT do the same thing as the Raspberry Pi ?
> The problem is that any way I look at gpio.cc I understand nothing. Do you have any [...]

Yes, the banana pi pins have been memory mapped at other addresses than in
the rpi, and controlling them needs different bits of code. One joy about
the banana pi is that only documentation about the pin mapping I found was
from the c-source of defunct python extension about controlling the pins
via sys/class/gpio, the pins are mapped on the gpio chip 1 with line
numbers being somewhere between 80 and 300. In other words trying to
actually use the gpio with a banana pi is quite a nightmare (so in line
with most their integrated stuff that just doesn't work properly).

GeorgeFlorian, 2019-07-07:
But should a Raspberry Pi 4 work ?

depili, 2019-07-07:
This is the source where you can extract the pin to register mapping
(which has no logic in it):
https://github.com/BPI-SINOVOIP/BPI-WiringPi/commit/bc19c3e86088f869627280b8f24bc9b1d4e326a8#diff-672071e4cd5d0146a80fb586c87c38e1R623

GeorgeFlorian, 2019-07-07:
> This is the source where you can extract the pin to register mapping (which has no logic in it): [BPI-SINOVOIP/BPI-WiringPi@bc19c3e#diff-672071e4cd5d0146a80fb586c87c38e1R623](https://github.com/BPI-SINOVOIP/BPI-WiringPi/commit/bc19c3e86088f869627280b8f24bc9b1d4e326a8#diff-672071e4cd5d0146a80fb586c87c38e1R623)

To be completely honest with you, I don't understand a thing from that.

hzeller, 2019-07-07:
yes, Raspberry Pi4 works, I have added support for it yesterday, @GeorgeFlorian