hzeller/rpi-rgb-led-matrix #835: Raspberry Pi 3 Model B: Led Matrix Displaying Nothing

Grudz, 2019-06-18:
The original and working system I was given came with a Raspberry Pi Model 3 B+ with 5 Led RGB Matricies attached to it. (These matrices are from Alibaba, the company is called LINSN ledstudio) It came with an Adafruit Bonnet with pins 4 and 18 soldered. The size of the Matricies are 32x64. This Pi was created based on the hzeller github.

I am trying to replicate this system using a different Raspberry Pi. In other words, I am using all of the original components, except a different Raspberry Pi which is a Model 3 B. I think I have followed all the instructions on this GitHub. However, when I try to run a command to display something to my Led's. The command "works" but no image is displayed. Let me explain...

Here are the commands I used, in this order:

git clone https://github.com/hzeller/rpi-rgb-led-matrix
          ***I also tried this entire sequence with ".git" at the end

sudo apt-get update

sudo apt-get install libgraphicsmagick++-dev libwebp-dev -y

make led-image-viewer
     ***I had to navigate to the "rpi-rgb-led-matrix/utils" folder to run this command

After this, I ran a file called rgb-matrix.sh that I found on the original Pi, the Model 3 B+ one, in the rpi-rgb-led-matrix folder. This file gave me the option to configure my Bonnet. When I ran this, I selected Bonnet, not a Hat +RTC. Also I chose the quality setting instead of the convience setting. (Remember, pins 4 and 18 are already soldered)

Finally, I ran the command in the image attached to this post. The "Hello World" example should have been displayed, however my led matrices remain completely dark.

P.S. If I put the original Model 3 B+'s SD card in my Model 3 B, an image displays and everything works.

[image]

GeorgeFlorian, 2019-06-19:
I'm having the same issue when trying to move from Raspberry Pi Zero to Banana Pi M3. The project works perfectly fine on the Pi Zero, but I get no output on the Banana Pi M3.

I'm starting to think that this library won't work on all of Raspberry Pi family devices but I haven't figured out why.
My main bet is the kernel version and/or the OS version.

The Pi Zero has and the library works:
```
Linux raspberrypi 4.19.42+
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
```
The Banana Pi M3 has and it does not work:
```
Linux bpi-iot-ros-ai 3.4.39-BPI-M3-Kernel
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
```

Grudz, 2019-06-20:
Interesting, it appears the OS that doesn't work for me is the one that's working for you. This is the command I ran:

uname -a
**Output** Linux Raspberry 4.19.42-v7+ #1219 SMP [date] armv71 GNU/Linux

Grudz, 2019-07-03:
It worked after I got new led-matricies for some reason

jaygarcia, 2019-08-01:
@Grudz please close this issue if you've resolved your problem.

closed 2019-08-01 by hzeller