hzeller/rpi-rgb-led-matrix #837: Top half of panels in 3rd chain has problems with blue

d3d9, 2019-06-25:
I use the Raspberry Pi 3 Model B+ (also tested this on a Pi Zero W, same result) with the "RGB Matrix Panel Drive Board For Raspberry Pi" from Electrodragon.
The top half of both panels in the 3rd chain has problems showing the blue color.

A full blue color like in the square image demo or in the demo that lights up different colors and transitions between them does not light up at all. In between, with the less bright blue colors, there is some flickering of the blue colors, they light up, but not really like they should.

Here are some pictures (of the two panels in the third chain):

The third line has the correct color behind the text, it's rgb(0, 8, 9).
[image]

Here's a closeup of the left part of the top 2 lines. Some of the blue lights actually light up, you can see the way it happens in this __[video](https://photos.app.goo.gl/99a4xpvQ2DG5moWa7)__.
[image]

A closeup of the orange text (in the area without the blue background). The color of the text is rgb(255, 65, 0), it should not show any blue, but it is visible on some pixels.
[image]

Here is the bottom left part of the test square, like in the demo with the color transitions, the "more than a little" blue is not visible at all:
[image]

d3d9, 2019-06-27:
A video of the other demo: <https://photos.app.goo.gl/57DFucXrPKHhjD9W6>
There was no "flickering" of the light blue color today (maybe because I actually connected all 3 chains), the main problem still persists though.

hzeller, 2019-07-07:
I suspect you need to inspect your hardware if the blue line is properly routed through. Either the problem is on the adapter board or the wiring to it.

hzeller, 2019-07-17:
Also, if you have problems with long chains, try a higher gpio-slowdown.
Always make sure to have very short cables between the panels.
Make sure, power is stable.

hzeller, 2019-07-17:
Also, did you make sure to switch off I2C on your Pi (see troubleshooting section)? The blue line for the third panel uses that pin.

d3d9, 2019-07-17:
I think I did that. I'm going to look at everything again in the next week and I'll check the settings, the adapter board (and the wiring without it as well) and I will try to find where the problem comes from.

d3d9, 2019-07-22:
I did the wiring manually now (only connected the 16 pins for the 3rd chain using jumper cables) and it worked without problems, so it seems like it is something with the adapter board (it happens on at least two of them, not just a problem on one board).

After reconnecting the board and being about to write this reply, I tried toggling the switch it has (between I2C for the RTC; and the 3rd chain, but don't misinterpret this as me having forgotten to switch it to the 3rd chain the whole time :smile:) just to see what happens -- as expected, the result wasn't a very useful appearance, with the whole top half being blue -- but when switching back to the 3rd chain, it was kinda weird at first and after I just put my finger close to the "switch" without turning it back to I2C again suddenly the blue color on the 3rd chain top half worked as expected..

It would be interesting to know if someone else who owns this adapter board can experience similar things. In any case, it seems like this issue can be closed.

jbodswor, 2019-07-22:
I just bought two of these adapter boards and put it together last night using 3 parallel chains of 4 panels (64x32) using a Rasberry Pi 4.  I am only seeing red on the third chain.  Changing out the board for the other, changing out the Pi with another, and swapping the panels around confirmed that there is definitely a problem with the third chain on the board.

After extensively looking at the software and disabling the I2C and 1-wire interface, I am convinced that it is something to do with the board.  I plan to look more closely and use my multimeter to track the integrity of the pins.  I will report back when done.

BTW - are you using a Pi 4?  I am having trouble installing some packages with the minimal Buster Lite OS.  More specifically, not finding versions of some dependencies.  Are the maintainers a bit behind for Buster?  Output below when trying to install python-dev, libgraphicsmagick, libwebp-dev, and some others.

`sudo apt-get install libgraphicsmagick++-dev libwebp-dev -y

E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/b/bzip2/bzip2-doc_1.0.6-9_all.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libp/libpaper/libpaper1_1.1.26_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/b/bzip2/libbz2-dev_1.0.6-9_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libp/libpaper/libpaper-utils_1.1.26_armhf.deb  404  Not Found [IP: 93.93.128.193 80]`

hzeller, 2019-07-23:
not found messages usually means you need to `apt-get update` first to get the current list of available packages.

jbodswor, 2019-07-23:
UPDATE: Turns out that the switch was not seated correctly on both ElectroDragon boards.  Once the switch was seated correctly, everything works perfectly.  Must check that the switch is in the correct position.

jbodswor, 2019-07-23:
I have done the `apt-get update` and `apt-get upgrade`, however the files referred to in the package are not in the archive for Raspbian Buster.  For example, the file `http://raspbian.raspberrypi.org/raspbian/pool/main/b/bzip2/libbz2-dev_1.0.6-9_armhf.deb` doesn't exist and has been replaced by `http://raspbian.raspberrypi.org/raspbian/pool/main/b/bzip2/libbz2-dev_1.0.6-9.1_armhf.deb`.  I think that the maintainer of the package is yet to update the package for Raspbian Buster.

I am investigating how I can either make a request to the maintainer or manually install the ever so slightly newer version.

jbodswor, 2019-07-23:
SOLVED: Had to run `apt-get update --allow-releaseinfo-change` to allow the release of Buster to change from the version it shipped with to the latest one.  Found this advice here: https://superuser.com/questions/1456989/how-to-configure-apt-in-debian-buster-after-release

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828