hzeller/rpi-rgb-led-matrix #839: No image/inconsistent output on 64x64 panel

manfelt, 2019-06-26:
I bought this 64x64 panel from aliexpress a while ago:

[https://www.aliexpress.com/item/32650825019.html?spm=a2g0s.9042311.0.0.254133ed3j1CqW](url)

While the installation and wiring seems to be correct and attached according to the 64x64 panel instruction, my 32x16 panel works fine with the same settings:

[image]

 I'm having trouble showing any consistent picture at all on the 64x64 panel as shown here:

[image]

[image]

I use this line: `sudo examples-api-use/demo -D0`

hzeller, 2019-07-07:
This might be a panel that has a bit different controlling logic. Do you have a datasheet or know what chipset it uses ?

postelontwerpen, 2019-07-07:
@manfelt It's my guess but does this work?:
sudo examples-api-use/demo -D0 --led-gpio-mapping=regular --led-rows=64 --led-cols=64

NOTE: if you have an Adafruit Hat, use: --led-gpio-mapping=adafruit-hat-pwm instead of 'regular'

NOTE: I don't see an D pin on the panels you ordered... could be the problem. (the D line is needed for 32x32 > displays; 32x16 displays don't need it) Read this [https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md](url)

FYI: Here is my partlist:
[https://github.com/hzeller/rpi-rgb-led-matrix/issues/843#issuecomment-508990265](url)

jaygarcia, 2019-08-03:
Hi @postelontwerpen , the part list link doesn't open up anyhing :(

@manfelt were you able to get this project rolling?

postelontwerpen, 2019-08-03:
Hi

Im sorry. Did something wrong by pasting my links. Does it work when you copy the links and paste them in your browser?

jaygarcia, 2019-08-05:
Hi @postelontwerpen -- stupid me: Didn't think of even copying and pasting. :man_facepalming:  Thank you :D

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin