hzeller/rpi-rgb-led-matrix #840: RPI4:  library get stuck when running utils or demo

postelontwerpen, 2019-06-27:
Hello, 
As a great fan of this matrix-lib I'd like to run this on a Raspberrypi 4. But it get stuck when I start running the program.

When I run de matrix program on my Raspberrypi 3, it plays the Utils library great!
When I run the same program on my new Raspberry 4, it starts running without errors, but get stuck straigth away. I can not quit the script in my terminal anymore. Sound (bcm_2835) is disabled and I2C is enabled. I use the newest Raspbian Buster Lite version.

I also tried the regular demo. On the RPI3 it runs great, on the RPI4 it get stuck...
Any ideas? 
Thank you in advance!

d3d9, 2019-06-27:
It may be related to the crashes with the Raspberry Pi 3 Model A+, maybe you can try applying the fixes in #727 and/or #740.

postelontwerpen, 2019-06-28:
@d3d9 thank you for your quick reply. That gives me a direction. I let you know if  this is fixes my issue. Thanks!

hzeller, 2019-06-28:
I have an RPi4 on order, so I'll fix it as soon as I have one in my hands.

GeekyTim, 2019-06-28:
I have a Pi4.  I should have tested this months ago, but didn't get round
to it.

*Thanks*

*Tim Richardson*
*Twitter: @Geeky_Tim*

katze, 2019-07-06:
Hello,
same problem with mine Rpi4 using classic-pi1 wiring.
The bug does not seem to be related to Raspbian Buster because everything works well with the same SD card on Rpi3.

I am available if there is need to do additional tests.
Thanks

hzeller, 2019-07-06:
I just received my Pi4, so hopefully I can fix it today.

commit 7fdbf1ed (2019-07-06): Update Pi Model detection; allow degraded non-root operation.

o Update PI model detection. Now detects Raspberry Pi 4 (fixes #840)
  and Rasberry Pi 3 A+ (fixes #727)
o If we can't mmap /dev/mem, fall back to /dev/gpiomem. Unfortunately,
  this will not allow to read the 1Mhz timer nor allow for PWM to work
  properly (fixes #680)
closed 2019-07-06 by hzeller