hzeller/rpi-rgb-led-matrix #841: Matrix Bonnet does not work video-viewer?

sguzmanolmos, 2019-06-29:
Hi Guys, I've been trying to use the video viewer with my Raspberry Pi 3B, the Matrix Bonnet and two 64x64 2.5 mm pitch. But the led-image-viewer and video-viewer don't seem to work, only the demos are displaying, and in the GPIO options the bonnet is not featured. Is this board not supported, or am I doing something wrong? Thanks for your time!

hzeller, 2019-06-29:
the bonnet should be the same hardware mapping as the adafruit hat.

sguzmanolmos, 2019-06-29:
Thanks! When the Bonnet is been modified to control 64x64 2.5mm pitch panels it uses the `hat-pwm` Problem solved :)

sguzmanolmos, 2019-06-29:
Is it possible to mirror the Pi's HDMI output with the video-viewer?

hzeller, 2019-07-07:
Adafruit built a program that uses this library and replicates the framebuffer content https://github.com/adafruit/rpi-fb-matrix.git

closed 2019-07-07 by hzeller