hzeller/rpi-rgb-led-matrix #842: Emoji?

oculos, 2019-06-29:


oculos, 2019-06-29:
Hi!

What a wonderful library! Thank you very much!
I was wondering, how can I scroll text with Unicode/emoji characters? 
I tried some Norwegian letters using the python samples, and it was a no go. It would also be cool to scroll emojis. 

Best,

Francis

DenisShelema, 2019-07-03:
You can create your own font with emoji and Norwegian letters or convert existing TTF font to BDF format.

oculos, 2019-07-03:
I meant a full color emoji.

hzeller, 2019-07-06:
The current font-system just does bitmap fonts with binary colors, so no full-color emojis.
But I am happy to accept pull requests that does support these kind of fonts.

hzeller, 2019-07-17:
Regular unicode however should work fine (BDF only supports the basic multilingual plane, but that should be sufficient for most). Make sure to provide it in UTF-8 (and your font contains the glyphs you are interested in).

Of course, you can implement multicolor fonts yourself easily, but it is not included here as it is out of scope. It is easier if you use existing libraries to decode that and use rpi-rgb-led-matrix as canvas.

closed 2019-07-17 by hzeller