hzeller/rpi-rgb-led-matrix #843: I am going to buy led matrix device.

andrei0807, 2019-07-02:
I live in china.
I am going to led matrix device in china.
https://www.adafruit.com/product/3211
But I can not find this product in taobao or Alibaba.
Can you send me product's link.
Thanks.

hzeller, 2019-07-07:
If you look for HUB75 matrixes, you probably find something.

andrei0807, 2019-07-07:
@hzeller
I did not find device.
 Can you send me link?

hzeller, 2019-07-07:
Search for "hub75" on aliexpress for instance. You'll find tons of LED matrices.

postelontwerpen, 2019-07-07:
here you go:
P2.5 Indoor SMD2121 Full Color Led Display Module 1/32 Scan 320x160mm
[https://www.aliexpress.com/item/P2-5-Indoor-SMD2121-Full-Color-Led-Display-Module-1-32-Scan-320x160mm-Super-High-Definition/32845686589.html?spm=a2g0s.9042311.0.0.3da24c4dC5F3eD
](url)

Power Supply Transformers 5V 400W
[https://www.aliexpress.com/item/New-5V-2A-3A-4A-5A-6A-8A-10A-12A-20A-30A-40A-60A-SwitchLED-Power/32641542774.html?spm=a2g0s.9042311.0.0.3da24c4dC5F3eD
](url)

Flat Ribbon Data Cable For LED Display 16P 20cm
[https://www.aliexpress.com/item/5Pcs-FC-10P-14P-16P-20P-26P-34P-40P-Gray-Flat-Ribbon-Data-Cable-For-LED/32849047343.html?spm=a2g0s.9042311.0.0.3da24c4dC5F3eD](url)

RGB Matrix Panel Drive Board For Raspberry Pi
[https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/](url)

25V 3300UF 25 V / 3300 UF Aluminum Electrolytic Capacitors
[https://www.aliexpress.com/item/20PCS-25V-3300UF-25-V-3300-UF-Aluminum-Electrolytic-Capacitors-Size-16-25-3300UF-25V-Electrolytic/32877266431.html?spm=2114.search0204.3.34.59f67586Td48bN&ws_ab_test=searchweb0_0,searchweb201602_3_10065_10068_318_319_5727315_10696_10084_10083_10618_452_10304_10307_10820_532_10821_10301_5727215_204_10843_10059_10884_10887_100031_320_10103,searchweb201603_1,ppcSwitch_0_ppcChannel&algo_expid=fea176a5-28a5-4cbc-806a-5759c7aa1d81-5&algo_pvid=fea176a5-28a5-4cbc-806a-5759c7aa1d81&priceBeautifyAB=0](url)

andrei0807, 2019-07-07:
@partofthething 
Thanks for your response.
https://www.adafruit.com/product/3211
I want this device in aliexpress.

postelontwerpen, 2019-07-07:
Yeah, be carefull with this. Adafruit uses an own ledpanel (other pin-addressing) that could be different from those you referred to at Aliexpress. There are others, here in this Issue-list, that tried and failed. I did not build de Adafruit version myself. At least, I tested succesfully with the Adafruit Hat and two panels I described (url) in my previous reaction.

Just to be more save with Aliexpress-panels : build it with the description you can find in this Repo, instead of Adafruits. And use a RPi3 B+ or lower for now. Or buy the Adafruit stuff and build that one.

jaygarcia, 2019-08-02:
> @partofthething
> Thanks for your response.
> https://www.adafruit.com/product/3211
> I want this device in aliexpress.

@andrei0807 if  you're using a general raspberry pi, i suggest the electrodragon daughter board.

https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/