hzeller/rpi-rgb-led-matrix #844: Test between RPi3 and RPi4 after fix #840 gives worse results

postelontwerpen, 2019-07-07:
@hzeller At first: Thank you for your help and quick update of your library!! Hopefully I can contribute a little bit with my own tests to compare the output of a RPi3 and RPi4. So that it results in a updated version of this great library.

For all tests I used:
- the same SD-card with same latest Raspbian OS
- two versions of this ledmatrix lib: latest rpi3 version and the updates one for RPi4
- 8x 128x64 ledpanels in two chains
- A 3xhub75 adapter

First test:
** RPi 3 B+ (see picture rpi3-pwm-lsb-50.png)**
command: 
`sudo ./led-image-viewer -f -w3 dentist_tools.jpg tree.jpg toothpaste.jpg growth.jpg kids_smile.jpg trees_sky.jpg --led-gpio-mapping=regular --led-chain=4 --led-parallel=2 --led-rows=64 --led-cols=128 --led-pixel-mapper="U-mapper" --led-scan-mode=1 -D 0 --led-pwm-lsb-nanoseconds=50`
Conclusion: Works great, with a little refreshrate jittering
[image]

Second test:
** RPi 4 (see picture rpi4-pwm-lsb-50)**
command: 
`sudo ./led-image-viewer -f -w3 dentist_tools.jpg tree.jpg toothpaste.jpg growth.jpg kids_smile.jpg trees_sky.jpg --led-gpio-mapping=regular --led-chain=4 --led-parallel=2 --led-rows=64 --led-cols=128 --led-pixel-mapper="U-mapper" --led-scan-mode=1 -D 0 --led-pwm-lsb-nanoseconds=50`
Conclusion: very disturbing image in a wrong arrangement of the panels
[image]
.png)

Third test:
** RPi 4 (see picture rpi4-pwm-lsb-600)**
command: 
`sudo ./led-image-viewer -f -w3 dentist_tools.jpg tree.jpg toothpaste.jpg growth.jpg kids_smile.jpg trees_sky.jpg --led-gpio-mapping=regular --led-chain=4 --led-parallel=2 --led-rows=64 --led-cols=128 --led-pixel-mapper="U-mapper" --led-scan-mode=1 -D 0 --led-pwm-lsb-nanoseconds=600`
Conclusion: Good arrangement of the panels but huge flicker because of the high chosen pwm-lsb-nanoseconds
[image]

So my questions are: 
**1. Are the above command-settings correct? 
2. What can I change to get a stable image on the RPi4?**

hzeller, 2019-07-07:
The Raspberry Pi 4 is much faster than the Pi3+, so it might be too fast for your displays. Try changing the settings for `--led-slowdown-gpio`

postelontwerpen, 2019-07-07:
@hzeller **--led-slowdown-gpio worked!**  In combi with a slower pwm-bits its a stable view. You are a hero!! Thank you!

FYI: Here my most solid command for a slidshow of 3 images: 

sudo ./led-image-viewer -C -f -w3 image1.jpg image2.jpg image3.jpg --led-gpio-mapping=regular --led-chain=4 --led-parallel=2 --led-rows=64 --led-cols=128 --led-pixel-mapper="U-mapper" --led-scan-mode=1 --led-pwm-lsb-nanoseconds=50 --led-slowdown-gpio=2 --led-pwm-bits=8 --led-show-refresh

Around: 57.5Hzmax: 18768usec

Let me know if you see room for improvement to make it a higher refreshrate without losing image quality (pwm-bits, lowers image quality in my example)

Thank you for your help and great work!

closed 2019-07-07 by postelontwerpen