hzeller/rpi-rgb-led-matrix #845: Buildroot integration

Rekoc, 2019-07-08:
I am trying to integrate this wonderful module in buildroot (2019.02), configured for a Raspberry 3B+. The idea is to send bitmap picture via Bluetooth and display them on a RGB display 48*48.
However, I have got a lot of difficulty to integrate the module to buildroot as I said. The module itself is fine but I can't fix the Python module and the led-image-viewer.cc.

First, the Python module, buildroot can't find the file "python.h" and when trying to compile led-image-viewer.cc, buildroot can't find 'GraphicsMagick++'.

Maybe someone here also use Buildroot and could help me with that ?

Here my Makefile:
`################################################################################
#
# rpi-rgb-led-matrix
#
################################################################################

RPI_RGB_LED_MATRIX_VERSION = 4c9a4036f44f173b783a0ba701138605f8b7d95c
RPI_RGB_LED_MATRIX_SITE = git://github.com/hzeller/rpi-rgb-led-matrix.git
RPI_RGB_LED_MATRIX_INSTALL_TARGET = YES
RPI_RGB_LED_MATRIX_ARCHITECTURE = rpi-arm
RPI_RGB_LED_MATRIX_SCONS_ENV = CC="$(TARGET_CC)" CXX="$(TARGET_CXX)"

define RPI_RGB_LED_MATRIX_BUILD_CMDS
	$(MAKE) $(RPI_RGB_LED_MATRIX_SCONS_ENV) -C $(@D)
	$(MAKE) $(RPI_RGB_LED_MATRIX_SCONS_ENV) -C $(@D)/bindings/python
	cp -r $(@D)/../imagemagick-7.0.7-39/Magick++/lib/* $(@D)/include
	cp -r $(@D)/../imagemagick-7.0.7-39/Magick++/* $(@D)/include
	cp -r $(@D)/../imagemagick-7.0.7-39/MagickCore $(@D)/include
	cp -r $(@D)/../imagemagick-7.0.7-39/MagickWand $(@D)/include
	$(MAKE) $(RPI_RGB_LED_MATRIX_SCONS_ENV) -C $(@D)/utils
endef

define RPI_RGB_LED_MATRIX_INSTALL_TARGET_CMDS
	$(INSTALL) -D -m 0755 $(@D)/examples-api-use/* $(TARGET_DIR)/usr/examples-api-use
	$(INSTALL) -D -m 0755 $(@D)/utils/* $(TARGET_DIR)/usr/examples-api-use
endef

$(eval $(generic-package))
`

All the 'cp' commands are here to copy the GraphicsMagick++ files required by the code.

hzeller, 2019-07-12:
Note that this uses Graphicsmagick, not imagemagick.

The other thing is, tht the Makefile invokes `GraphicsMagick++-config` to determine where the libraries and include files are. Maybe you need to manually set `MAGICK_LDFLAGS` and `MAGICK_CXXFLAGS` to tell the makefile where to find thiings.

I'll make it possible to read from an environment variable.

hzeller, 2019-07-17:
Haven't heard anything, I assume it works now ?

closed 2019-07-17 by hzeller

ffontaine, 2021-12-09:
FYI, the package has been added to buildroot with https://git.buildroot.net/buildroot/commit/?id=e821078031e14df69e5142f23e9cb7b5a1ea0e77