hzeller/rpi-rgb-led-matrix #846: Can't build for python on DietPi

MisterBiggs, 2019-07-10:
I am currently using DietPi v6.25.3 on a RaspberryPi 3B on a totally minimal install. The only packages installed are git, make, and the ones listed in the python binding instructions. When I run the command to make the library for python3 I get:
```bash
root@DietPi:~/rpi-rgb-led-matrix# make build-python PYTHON=python3
make -C ./lib
make[1]: Entering directory '/root/rpi-rgb-led-matrix/lib'
g++ -I../include -Wall -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -Wextra -Wno-unused-parameter -fno-exceptions -c -o gpio.o gpio.cc
make[1]: g++: Command not found
Makefile:183: recipe for target 'gpio.o' failed
make[1]: *** [gpio.o] Error 127
make[1]: Leaving directory '/root/rpi-rgb-led-matrix/lib'
Makefile:15: recipe for target 'lib/librgbmatrix.a' failed
```

I get similar issues for python2 and when trying the C++ way as well:

```bash
root@DietPi:~/rpi-rgb-led-matrix# make -C examples-api-use
make: Entering directory '/root/rpi-rgb-led-matrix/examples-api-use'
g++ -I../include -Wall -O3 -g -Wextra -Wno-unused-parameter -c -o demo-main.o demo-main.cc
make: g++: Command not found
Makefile:41: recipe for target 'demo-main.o' failed
make: *** [demo-main.o] Error 127
make: Leaving directory '/root/rpi-rgb-led-matrix/examples-api-use'
```

hzeller, 2019-07-10:
you don't have a g++. Did you install build-essentials?

MisterBiggs, 2019-07-11:
That was it. 
Running `sudo apt-get install build-essential` fixed the issue.

closed 2019-07-11 by MisterBiggs