hzeller/rpi-rgb-led-matrix #849: 80*40 new multiplex mapper

krishnasandeep10694, 2019-07-11:
Hi @hzeller,
Thanks for the awesome library. I have some doubts about multiplex mapping. I bought some 80*40 panels and tried this library on it. Pixels were disarranged so i wrote a new multiplex mapper looking at existing code of multiplex mappers to correctly arrange the pixels. 
[image]
I bought new bunch of led panels as the old ones are not available thinking that i can write a new multiplexing mapper but the content now is getting duplicated instead of disarranged as shown in the picture below.
[image]

Could you help me figure out what might be the reason.

I didn't understand how panel stretch factor is used. Could you explain it.

Thanks

hzeller, 2019-07-12:
This looks like it might have a different `--led-row-addr-type`

krishnasandeep10694, 2019-07-12:
Hi,
I have tried with other --led-row-addr-type options but the leds are not lighting up. I have read the wiring.md and assuming that wiring would be same for any row-addr-type. I am posting the hub 75 image i found on the back of the panel. 
[image]

Please have a look at it and suggest us how to proceed.

Thanks

hzeller, 2019-07-12:
This might also be a panel that does some sort of double-edge data pushing, but I don't have one of these. If you can reverse engineer this one with strategic pulse generation or finda a datasheet, we can probably do more here.

cPabz, 2021-11-21:
@krishnasandeep10694 I need some help connecting 80*40 displays, i have the same issue as you...
I am looking for help with the multiplex mapping...
Is there a way I can reach out to you, please share..

cPabz, 2021-12-23:
@krishnasandeep10694 @hzeller can either of you help me understand stretch factor..?
I have successfully written a multiple map for 80*40, but for some reason my lower half of image seems to be mirror image of upper half. Any help is appreciated.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828