hzeller/rpi-rgb-led-matrix #850: Has anyone tested with the Pi4?

wibbleboy, 2019-07-15:
I've been using a setup using a Pi3 Model B, the Adafruit Matrix Bonnet, and LED displays which I sourced from China. These have all been working fine, however I've just purchased a new Raspberry Pi 4 Model B (the 4GB model), and after swapping out the Pi3 for the Pi4, the LED panels no longer display anything.

I'm currently not sure if there's an incompatibility with the Pi4 at the hardware level, a problem with Raspian switching to Buster, or maybe even faulty GPIO pins on the new board.

Has anyone successfully run any of the samples against a Pi 4 Model B?

hzeller, 2019-07-15:
did you update to the latest version of the library?

wibbleboy, 2019-07-15:
That's a fair question. I'd pulled what I assumed was a fresh version using the scripts supplied by Adafruit:

[https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/rgb-matrix.sh](url)

I'd never noticed this before, but those scripts reference a specific commit (814b79b5696d32dd1140304b41a1ec0068bb271a) which is 9 months old. I'll try pulling in the latest version of the libraries to see if that helps.

hzeller, 2019-07-15:
can you file a bug with them asking to  fetch a recent version in their installer?

wibbleboy, 2019-07-15:
I've just successfully retested the Adafruit scripts by manually editing `rgb-matrix.sh` and replacing the existing commit with the latest revision before running it. This all seems to work fine afterwards. I'll try to push through a change on their end, now that I know that's the fix. Thanks for the help.

closed 2019-07-16 by wibbleboy

coulix, 2019-07-16:
I got mine today, I will look at it this week.

wibbleboy, 2019-07-16:
The fault I encountered was with the Adafruit scripts for their bonnet, not Henner's matrix library. I branched the Adafruit tutorial, manually edited the library reference, then tested it on the Pi 4. This all seemed to run fine on my Pi4.

I've raised a pull request to try and get this change into Adafruit's formal channel; but until then, the updated script can be run:

```
curl https://raw.githubusercontent.com/wibbleboy/Raspberry-Pi-Installer-Scripts/patch-1/rgb-matrix.sh >rgb-matrix.sh
sudo bash rgb-matrix.sh
```