hzeller/rpi-rgb-led-matrix #852: Steady flickering issue

mansj, 2019-07-16:
I've got a flickering issue using the Python bindings. I've looked over all the suggestions posted elsewhere (isolating CPU core, pwm-settings, soldering adafruit pins, etc) but it won't go away.

Here are some details:

• Flickering is not visible using the supplied demoes (examples-api-use/demo)
• Flickering is not visible when using the Python bindings and not doing much else than updating the matrix

The current options I'm using are

```
options = RGBMatrixOptions()
options.rows = 16
options.cols = 32
options.chain_length = 8
options.gpio_slowdown = 2
options.multiplexing=3
options.brightness=70
options.hardware_mapping="adafruit-hat"
options.pixel_mapper_config="rotate:-90"
```

Since I can get the Python bindings to work without flickering if I don't add a bunch of code to my Python game, it seems to me that it would be a CPU issue, i.e. if the Python code is "too demanding", there is no time left to keep feeding data to the matrix. But looking at the output from "top" I can't see any throttling.

Is there a way to determine definitely _where_ the bottleneck is?

hzeller, 2019-07-17:
After the usual suggestions (simplest Raspberry Pi installation, e.g. DietPi, no GUI, switching off service you don't need etc.) which looks like you already did, the PWM modification is the next important thing, otherwise you _do_ will see flicker.

So If you use the Adafruit Hat (which it looks like from your settings), you really want to use the PWM variant, by doing the very simple hardware modification: https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker
Otherwise you'll have fluctuating brightness flicker etc. Then choose
```
options.hardware_mapping="adafruit-hat-pwm"
```
If you do the hardware modification, it won't make a difference until you choose `adafruit-hat-pwm`.

One other problem is: You have a fairly large chain length, so you would benefit from driving these panels in parallel instead of a single chain, but it would mean to [solder your own adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3).

Python is pretty demanding to the memory subsystem (as it has to garbage collect), which will take away memory bandwidth; and the Pi is not very strong in that department. So even if it does not show up in the CPU, it might still be an issue. Try to make your code in a way that it does not require too much garbage collection, by re-using objects etc (or write it in C++, but I guess I am too late with that suggestion :) ...). Anything more complicated (e.g. image decoding) will result in flicker, so try to pre-calculate these things, e.g. by pre-filling frame-canvases etc.

If you have a situation in which the frame rate fluctuates, it [might help to fix the time it uses for updates](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L105).
Since it is a compile-time option, make sure to re-compile all the Python bindings as well.

mansj, 2019-07-23:
Thanks for your detailed comments! I've made the hardware mod, and it does look more stable after that, but I'm still having som flickering. 

I'm fairly certain I have narrowed it down to being something to do with my other Python code though, simply not leaving enough cycles for the matrix updates, so I'll work on refining that. Thank you!

closed 2019-07-23 by mansj