hzeller/rpi-rgb-led-matrix #853: Constant line artifact

danielvy, 2019-07-17:
I have a two 32x32 panels from Adafruit chained using the Adafruit hat and soldered the 4-18 pins as suggested. 

Everything was working for quite some time but recently, there is a constant solid line artifact appearing regardless of anything drawn. Just by initializing and clearing the display, this is what I get:

[image]

And if a set some red area on top of it (the blue line is still visible, but it is now mixed with the red):

[image]

Always happens on the left panel, also when I connect it directly (and not chained through the first one). It is not effected by refresh rates or slowing down the GPIO.

Sometimes, on rare occasions, when particular combinations of leds are lit, it briefly disappears but I can't reproduce this.

It started appearing after updating the led-matrix library but I am not sure if this is a software issue or a defect in the circuit itself. 

What can cause this and is there any way to isolate the problem?

Thanks.

hzeller, 2019-07-17:
this looks like a hardware problem on the panel.

davemaster, 2019-07-17:
Cheks this: swap panels and picture

danielvy, 2019-07-17:
> Cheks this: swap panels and picture 

Can't really swap them (they are glued to the frame and I don't have cables that long). 

But I can connect only the left one directly without chaining (the right one is disconnected). When I do that, I get the exact same blue line in the same place.

closed 2025-11-27 by marcmerlin