hzeller/rpi-rgb-led-matrix #854: Must run as root to be able to access /dev/mem when using together with Flask

jghaanstra, 2019-07-17:
I'm trying to built a simple WebAPI with Flask for controlling my LED matrix. Controlling the LED matrix with the Python API works just fine. But when I add the code for Flask to create the endpoints I get the following error.
```
 * Serving Flask app "python-led-matrix-api" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: on
 * Running on http://0.0.0.0:5050/ (Press CTRL+C to quit)
 * Restarting with stat
Must run as root to be able to access /dev/mem
Prepend 'sudo' to the command
```
I dont understand why using Flask together with rpi-rgb-led-matrix would generate this error. My code looks like this (beware, I'm a python newbie)

```
from flask import Flask, request, jsonify
from flask_restful import Resource, Api
from rgbmatrix import RGBMatrix, RGBMatrixOptions, graphics
import time

app = Flask(__name__)
api = Api(app)

options = RGBMatrixOptions()
options.hardware_mapping = 'adafruit-hat'
options.rows = 64
options.cols = 64
options.chain_length = 1
options.parallel = 1
options.row_address_type = 0
options.multiplexing = 0
options.pwm_bits = 11
options.brightness = 100
options.pwm_lsb_nanoseconds = 130
options.led_rgb_sequence = 'RGB'
options.gpio_slowdown = 1

matrix = RGBMatrix(options = options)
offscreen_canvas = matrix.CreateFrameCanvas()

# FUNCTIONS
def RunText(textfont, xpos, ypos, r, g, b, text):
    font = graphics.Font()
    font.LoadFont(textfont)
    textColor = graphics.Color(r, g, b)

    offscreen_canvas.Clear()
    len = graphics.DrawText(offscreen_canvas, font, xpos, ypos, textColor, text)
    xpos -= 1
    if (xpos + len < 0):
        xpos = offscreen_canvas.width

    time.sleep(0.05)
    offscreen_canvas = matrix.SwapOnVSync(offscreen_canvas)

def Fill(r, g, b):
    offscreen_canvas.Clear()
    offscreen_canvas.Fill(r, g, b)
    offscreen_canvas = matrix.SwapOnVSync(offscreen_canvas)

# ENDPOINTS
class runText(Resource):
    def post(self):
        json = request.get_json()
        #p = subprocess.Popen(['python', './library/runtext.py', '--led-rows', '64', '--led-cols', '64'])
        RunText(json['font'], json['position']['x'], json['position']['y'], json['color']['r'], json['color']['g'], json['color']['b'], json['text'])

        return jsonify({"success": True})

class fillColor(Resource):
    def post(self):
        json = request.get_json()
        #p = subprocess.Popen(['python', './library/fill.py', '--led-rows', '64', '--led-cols', '64', '-r', '255', '-g', '0', '-b', '0'])
        Fill(json['color']['r'], json['color']['g'], json['color']['b'])

        return jsonify({"success": True})

class Clear(Resource):
    def get(self):
        p = subprocess.Popen(['python', './library/clear.py', '--led-rows', '64', '--led-cols', '64'])

        return jsonify({"success": True})

api.add_resource(runText, '/text')
api.add_resource(fillColor, '/fill')
api.add_resource(Clear, '/clear')

if __name__ == "__main__":
    app.run(host="0.0.0.0", port=5050, debug=True)
```

hzeller, 2019-07-17:
You always have to run as root to generate proper images ( https://github.com/hzeller/rpi-rgb-led-matrix#running-as-root )

That being said, since recently (a couple of days ago), it it also possible to run without root, but it will be degraded performance, so it is still not desired.

jghaanstra, 2019-07-17:
Thanx for the quick response @hzeller . I forgot to mention that I'm running as root already.

```
pi@raspberrypi3bplus:~/python-led-matrix $ sudo su
root@raspberrypi3bplus:/home/pi/python-led-matrix# sudo python python-led-matrix-api.py
 * Serving Flask app "python-led-matrix-api" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: on
 * Running on http://0.0.0.0:5050/ (Press CTRL+C to quit)
 * Restarting with stat
Must run as root to be able to access /dev/mem
Prepend 'sudo' to the command
root@raspberrypi3bplus:/home/pi/python-led-matrix#
```

hzeller, 2019-07-17:
did you compile from a recent version if the library? Hardware detection of the Pi3+ was not working before unless you manually edited the code.

hzeller, 2019-07-17:
do you drop privileges anywhere in your flask app before initializing the rgb matrix? try starting it as early as possible, _before_ starting flask, which might do privilege dropping.

jghaanstra, 2019-07-18:
I used the outdated script from Adafruit but adjusted it to the latest commit so it should be pretty up to date.

But I left for holidays this morning. So I will check in three weeks ago again and post back here.

angelogoncalve, 2019-07-18:
Add this options when you are creating the list of options that have the matrix/panel/display:

options.drop_privileges = 0
options.daemon = 0

jghaanstra, 2019-08-07:
> Add this options when you are creating the list of options that have the matrix/panel/display:
> 
> options.drop_privileges = 0
> options.daemon = 0

Thanx, this did the trick!

closed 2019-08-07 by jghaanstra