hzeller/rpi-rgb-led-matrix #855: Problem with FM6126A chipset

z26100, 2019-07-21:
Hi,
I bought some of [these](https://www.ebay.de/itm/4x-P4-RGB-LED-Matrix-Panels-Display-Austauschmodule-64x64-pixel-256-x-256-mm/303174069343?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649) panels and after I applied the [reset_active3.py.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/3139328/resetmatrix_active3.py.txt) I could at least see them working in general.
[image]
[image]

[image]

However, the second half of the panel seems not to be addressed properly. 
After some investigations I think they consist out of 2x32 rows that are some chained together, so I also tried out
```
 sudo ./demo --led-no-hardware-pulse --led-gpio-mapping=regular --led-rows=64 --led-chain=2 -D 0
``` 
which leads to one single rotating cube but still the same LED rows remain blank/black.
[image]

Any suggestions?

z26100, 2019-07-21:
After carefully reading the [documentation](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) of the active board in this repository it seems I've missed to solder the bridge for addressing the E line.

sachasayan, 2019-07-31:
I've added a hardware compatibility table in the Wiki, could you add notes for this panel there with settings if you've confirmed it working? ^_^

https://github.com/hzeller/rpi-rgb-led-matrix/wiki

hzeller, 2019-09-03:
Looks like you got it working with connecting the E line.

closed 2019-09-03 by hzeller

marcmerlin, 2019-12-23:
Hi @hzeller , so it is mostly working, but not quite.
For a setup with more than 128 columns, the init sequence does not get sent deep enough. See https://github.com/hzeller/rpi-rgb-led-matrix/issues/947 for details

marcmerlin, 2019-12-30:
whooops, ignore my comment on the wrong bug, I'll delete it.