hzeller/rpi-rgb-led-matrix #856: Clearing an area, not the complete screen

ChristopherBadenhorst, 2019-07-22:
Hi,

I'm making a clock with some other features. Now I want the clock to update every x time, and that works fine. I just can't figure out how to clear only the clock area and not the whole screen with every update of the clock. 

I found: 
`matrix = RGBMatrix(options = options) `
and 
`matrix.Clear()`

I do think I can use canvas.Clear() for just an area. I just don't understand how I can define the 'canvas' area.
Can anyone help?

angelogoncalve, 2019-07-26:
x times y is the area/matrix to clear:

 
matrix.SetPixel(x, y, 0, 0, 0)

ChristopherBadenhorst, 2019-07-26:
@angelogoncalve 
Thanks for your help, I've tried a lot of variations with your piece of code but can't work it out.
What is wrong here now? 

`def job2():
    for y in range(0, 32):
    for x in range(0, 64):    
    mseconds.SetPixel(0, 32, 0, 0, 0)
    
    mseconds.Clear()
	
    clock = datetime.datetime.now().strftime("%H:%M:%S")
    graphics.DrawText(matrix, font3, 0, 55, blue, clock)
        
job2()
schedule.every(1).seconds.do(job2)`

ChristopherBadenhorst, 2019-07-26:
I do still want to rebuild it to just clear the area of the 'seconds', and make a seperate job for the minutes and a seperate job for the hours. But if I get this right, i'll get all of them right.

hzeller, 2019-07-26:
I suppse you want to casll `SetPixel()` with x and y, no only 0, 32 ?.

Anyway, what you should do is to write the time into a second canvas, then do `SwapOnVSync()`, otherwise this will result in flicker.

angelogoncalve, 2019-07-27:
Yes Henner Zeller you arer correct:

def job2():
    for y in range(0, 32):
    for x in range(0, 64):    
    mseconds.SetPixel(x, y, 0, 0, 0)

ChristopherBadenhorst, 2019-07-27:
The SwapOnVSync() made it more complicating for me. Solved it (quick and dirty) with drawing an image that is completely black, before writing the new time. 
Works very wel!

jaygarcia, 2019-08-02:
@ChristopherBadenhorst, if you've solved this, please consider closing this issue =)

closed 2019-08-02 by ChristopherBadenhorst