hzeller/rpi-rgb-led-matrix #859: Small pitch 64x64 matrix

mrnick1234567, 2019-07-29:
Hey Henner, you very kindly contributed to some of my projects a while back. I was looking at your library and the following 1/32 scan hub75e matrix:

https://s.click.aliexpress.com/e/c6XiGcd6

Could you tell me if you think it would work with a pi and the adafruit bonnet (with the solder change to enable the 5bit data / E pin)

I see a comment in one of the reviews for this matrix that I should be sure to ask for ine using 'Chip One ICN2037' if I'm using raspberry pi. Does that sound right?

Thanks!

GeekyTim, 2019-07-29:
I have those, and they work together really well.

mrnick1234567, 2019-07-29:
Thanks Tim! What chip is on them, is it the ICN2037? Just want to be sure I order the right thing.

GeekyTim, 2019-07-30:
I'll take a look when I get home next week.

amantalion, 2019-12-27:
@GeekyTim Did you confirm that you have ICN2037? Thank you so much for your help by the way. Really helps to know that it is working before placing an order.

AndreasMainz, 2020-06-12:
Hello,
I am using 64x64 Panel with ICN2037 and GM7258 and it works fine with 
Joy-it LED Matrix controller out of the box.
Command is: 
demo -D 0 --led-rows=64 --led-cols=64 --led-slowdown-gpio=2
Also 2 panels chained works fine:
demo -D 0 --led-rows=64 --led-cols=128 --led-slowdown-gpio=2
Maybe this helps somebody to setup.