hzeller/rpi-rgb-led-matrix #860: Calculation of number of LEDs

hemanthkd, 2019-07-30:
Hi there, I wanted to understand how can we calculate the total number of LEDs that can be controlled by one single raspberry pi to play a video without any flickering effect

jaygarcia, 2019-07-30:
@hemanthkd i imagine part of it has to do with what device, if you've configured `isolcpus=3` in `/boot/config.txt` and how many pam-bits (number of colors?) you're willing to deal with.

I can tell you that I've got a 3b+ controlling 4x 64x64 (16,384 pixels) panels in serial with no issues. 

This screen has 5 Pi's running to make full-motion video via TCP/IP. Each Pi drives 16,384 pixels exactly.  To drive this many pixels > 60fps, I had to use `pwm_dither_bits=2` and `pwm_bits=4`, which reduced colors a little.

Demo:
https://www.youtube.com/watch?v=s0lbVMcgfww&t=13s

hemanthkd, 2019-07-31:
I have done the configuration at /boot/config.txt, still, we are facing flickering & other disturbances, we are running it in 2 chains, a total of 160*160LEDs(total of 8 panels in 2 chains & each chain using U mapper ) by one pi 
Each Panel: 40 in rows & 80 in cols

Also, can you help me to do the multiple pi synchronization?
Thank @jaygarcia

angelogoncalve, 2019-08-01:
Device you're using to drive the display (Pi2, 3, 3b+). Also RPi4 have more processor.

jaygarcia, 2019-08-01:
@hemanthkd I'm curious as to why you're using the U Mapper?

To help, We'll need:

- Pi model
- What OS
- Are you actively / passively cooling your Pi?
- What exactly you're running on the Pi (demo via compiled code or python?)
- Any other software running concurrently
- Specs for your panels  (model, refresh rate, port, power)
- How you wired your panel to the pi (daughter board, direct, etc..)

For the Pi4's CPU capabilities, I was able to drive ten 10 (64x64) panels and got results around 70-75hz using the following command line: 

```./demo  -D 9 --led-rows=64 --led-cols=64 --led-parallel=2 --led-pwm-bits=9 --led-slowdown-gpio=3 --led-chain=5 --led-show-refresh```

angelogoncalve, 2019-08-02:
Operating System for RPi: Diet Pi

hemanthkd, 2019-08-03:
@jaygarcia 
Here are the details of the configuration that I am using:

- Pi model: Pi 3 Model B+
- What OS: 2019-04-08-raspbian-stretch
- Are you actively / passively cooling your Pi?: I haven't added any heatsinks other than the metal shield which is already present
- What exactly you're running on the Pi: Playing a video using the compiled video player code
- Any other software running concurrently: remote desktop
- Specs for your panels: 
  Model: Q4 Indoor
  Refresh Rate:>1920Hz
  Power=24W
- How you wired your panel to the pi: active -3 Board

@angelogoncalve: Thanks for the suggestion

jaygarcia, 2019-08-03:
@hemanthkd you might be doing too much on your Pis, and I strongly suggest you put some active cooling on them.  To run the RGB Matrices, Core 4 will likely spike to > 80%, causing the CPU temp to rise. Once it hits its limit, it throttles down, causing flicker.

I've been successful with a few projects:
My RGB Screen uses 5x Pi3B+ to run each vertical "segment" of 64x64 rgb panels. 

Here's a demo of it working: https://www.youtube.com/watch?v=s0lbVMcgfww

The raspberry Pi 4 demo is here:
https://www.youtube.com/watch?v=1aDurOOmkeo

Apologies for mispronouncing @hzeller  name as "Hennen".

hemanthkd, 2019-08-07:
@jaygarcia  Yes, it is pi3B+ is hitting a temperature of 56.9'C to 58'C

jaygarcia, 2019-08-09:
@hemanthkd if you want to really get the most out of your Pi3B+, change the OS to DietPi and ensure `isolcpus=3` is in `/boot/cmdline.txt`.

Running the desktop environment is a ton of overhead.

hemanthkd, 2019-08-10:
@jaygarcia thanks for the update, I will test with the diet pi, I wanted to understand how to synch multiple pis to play a single video in a large video wall

closed 2025-11-27 by marcmerlin