hzeller/rpi-rgb-led-matrix #861: Raspberry Pi 4 -- has anyone tried it yet?

jaygarcia, 2019-07-30:
Just got a Pi4 in and am wondering if anyone's had a chance to try it out w/ this lib? Seems that basic demos that run well on the 3b+ are freezing, requiring me to SSH in and `kill -9 <pid>` to cancel execution.

=D

jaygarcia, 2019-07-30:
Seems that i had to update my library to get this running. However, there is an issue (line noise?) preventing a clean display.

Swap out this setup for a 3b+ and it's perfect :D

Specs:
- [ ] RPI 4
- [ ] Updated version of Raspbian Stretch
- [ ] [Electrodragon RGB Matrix drive board](https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/)
- [ ] 64x64 P2.5 RGB Panels @ 32s (`P2.5-2121-32S-V1`)
- [ ] Command executed: `./demo  -D 9 --led-rows=64 --led-cols=64  --led-parallel=1 --led-pwm-dither-bits=2 --led-pwm-bits=4`

__Demo video below__
[image]

hzeller, 2019-07-30:
increase the gpio slowdown value, the pi4 is too fast for most panels.

jaygarcia, 2019-07-30:
> increase the gpio slowdown value, the pi4 is too fast for most panels.

D: too fast? I had no idea that would be a thing. I'll hit that now and circle back.

Thanks @hzeller =D

jaygarcia, 2019-07-30:
confirming `--led-slowdown-gpio=2` works perfectly :)

closed 2019-07-30 by jaygarcia

DeepeshKumarKushwaha, 2019-08-01:
On RPi 4 When I display text then flickering (noisy text ) is there. When I went through lib it is mentioned that --led-slowdown-gpio values can be 0 to 4 but unable to do 3 or 4.I tried with slowdown-gpio values from 0 to 2 their is a lot of noise.

Command:- sudo python uart_rgb.py  --led-rows=32 --led-cols=32 --led-chain=4 --led-parallel=2 --led-multiplexing=1 --led-brightness=100 --led-pwm-bits=4  --led-slowdown-gpio=4 --led-scan-mode=1 --led-rgb-sequence=RGB

Error:-usage: uart_rgb.py [-h] [-r LED_ROWS] [--led-cols LED_COLS] [-c LED_CHAIN]
                   [-P LED_PARALLEL] [-p LED_PWM_BITS] [-b LED_BRIGHTNESS]
                   [-m {regular,adafruit-hat,adafruit-hat-pwm}]
                   [--led-scan-mode {0,1}]
                   [--led-pwm-lsb-nanoseconds LED_PWM_LSB_NANOSECONDS]
                   [--led-show-refresh] [--led-slowdown-gpio {0,1,2}]
                   [--led-no-hardware-pulse LED_NO_HARDWARE_PULSE]
                   [--led-rgb-sequence LED_RGB_SEQUENCE]
                   [--led-pixel-mapper LED_PIXEL_MAPPER]
                   [--led-row-addr-type {0,1,2}]
                   [--led-multiplexing LED_MULTIPLEXING] [-t TEXT] [-i IMAGE]
uart_rgb.py: error: argument --led-slowdown-gpio: invalid choice: 3 (choose from 0, 1, 2)

hzeller, 2019-08-01:
The range of gpio slowdown is double-checked by the python args parser, which it shouldn't. If you are using samplebase.py, this is fixed in 
https://github.com/hzeller/rpi-rgb-led-matrix/commit/5e004a96d94112ca5c76b733702e5458baacf70e

DeepeshKumarKushwaha, 2019-08-02:
yes now working fine
Thanks @hzeller