hzeller/rpi-rgb-led-matrix #863: garbage pixels altough not constantly

flok99, 2019-08-03:


flok99, 2019-08-03:
Situation:
- 8 panels, 32x16 each, adafruit 420
- brightness 50%
- --led-rows=16 --led-chain=8 --led-pixel-mapper="U-mapper"  --led-no-drop-privs
- adafruit hat, PWM patch applied, software compiled with HARDWARE_DESC?=adafruit-hat-pwm
- two powersupplies (gnd connected)
- raspberry pi 4 with 4.19.57-v7l+ kernel

With 4 panels (--led-chain=4) the background is black but scrolling text is noisy on the edges.
With 8 panels, the background is all noise, text shows fine altough with the same noise on the edges.
I went through the help-page (disabled all i/o like i2c, spi, serial, audio, etc).

Any ideas? Please see attached photos.

[image]
[image]

hzeller, 2019-08-03:
The Pi4 is typically too fast for the panels. You typically want to set the gpio slowdown to a higher value.

buzztiaan, 2019-08-03:
i just threw a pi3 at his desk, gonna try in a bit if that fixes it

flok99, 2019-08-03:
I've tried --led-slowdown-gpio=4 and this helps dramatically; no noisy edges and it doesn't flicker anymore. The color-patterns stay though.

Next step is testing with the rpi3.

flok99, 2019-08-03:
With the RPI-3b we have the same problem.

hzeller, 2019-08-03:
Could also be high inductance in your power cables. Keep them as short as possible and consider adding capacitors directly to the power connection.

(Weird patterns often happen if 1-wire interface is switched on, but you said you already took care of disabling interfaces).

flok99, 2019-08-03:
Ok, closing the ticket.
The problem was that the rpi4 was too fast and...a bug in our software :-(

flok99, 2019-08-03:
thanks for your help by the way!

closed 2019-08-03 by flok99