hzeller/rpi-rgb-led-matrix #864: 64x64 Matrix + Adafruit Bonnet blank rows

Verrou, 2019-08-04:
Hi there,

Thank you so much for this great library, I'm somewhat new to all this so please forgive me if I've made a rookie error!

When running any script to my matrix, rows 17-32, and 49-64 do not run/are blank? With the D10 demo, some of those blank LEDs do emit a very faint light, but otherwise nothing.

I'm running a RPi4 with the Adafruit Bonnet and [this matrix from AliExpress](https://www.aliexpress.com/item/32733178058.html). I have 2 panels and both have the same result.
[image]
[image]

Any help is appreciated, many thanks!

hzeller, 2019-08-04:
if you have a 64x64 matrix you need to change add an E-line: https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet

jaygarcia, 2019-08-05:
@Verrou suggest you test other demos once you get your Line E soldered up.  I found that the Pi4 produced noise and posted this issue: #861

Henner was quick to point out that I needed to use a flag like `--led-slowdown-gpio=2` to clean things up and it worked like a champ.

Verrou, 2019-08-06:
Hmmm the bonnet comes with a wee connection to easily do this, but I tried both 8 & 16 with no luck. Might just have to buy an Active3 adapter. Does anyone sell them totally finished? 
[image]

jaygarcia, 2019-08-06:
I suggest the Electrodragon board: https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/

Also, I wonder if it makes a difference to ground Pin 4?  The elctrodragon board has pin holes that you jumper to select P4 > GND and P8 to E line.

Verrou, 2019-08-07:
Okay, so I think I have tried all the solder points under the sun haha!

I've tried P4 > GND as above, but that just stopped it working all together. So tried P4 > P18 as per one of Adafruit's projects (the recommend installing a jumper between these two for 64x matrices). And the combination of that and E > P8, has resulted in the blank LEDs turning on! Essentially moving the bars up/down. At least I know all the LEDs work?
[image]
So frustrating because I feel like I'm so close, yet so far away!!!
p.s. I really appreciate your help on this :)

jaygarcia, 2019-08-09:
@Verrou One thing to do is wire the Panel directly to the screen per the instructions in this repo. You won't have the 3 to 5V level shifters/buffers.  You can at least use this method to ensure that the adafruit bonnet is not the cause.

angelogoncalve, 2019-08-10:
Try theses flags:

--led-rows=64 --led-cols=64 --led-row-addr-type=1
or
--led-rows=64 --led-cols=64

carlosalaniz, 2019-10-20:
I'm having the same issue, directly connected to the pi. @Verrou Were you able to solve this?

[image]

[image]

FredericoCoelhoNunes, 2021-01-03:
Exactly the same problem here. My 64x64 matrix is from the same manufacturer, and I have the same Adafruit bonnet, but soldering to 8 or 16 didn't help. Should have just dished out a bit more money and bought matrices from Adafruit instead of random stuff from Alibaba. Oh well!

If anyone finds a solution, I'd love to hear about it. Cheers.

hzeller, 2021-01-03:
If you need 64x64 with the Adafruit bonnet or hat, you must follow the [instructions to add the E line](https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet) as they don't connect an E addressing.

Alternatively, build an [adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter) or [get a 3rd party one](https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/) that already has this covered.

jaygarcia, 2021-01-03:
The electro dragon board has been very good to me. 🥳

angelogoncalve, 2021-01-03:
another boards:

https://www.adafruit.com/product/2601

https://www.adafruit.com/product/3036

mabanham, 2021-06-20:
Did anyone find a solution to this? I have had the same issue using the Adafruit matrix bonnet with the 64x64 1/32 displays (Hub75E pinouts and FM6126A chips). I've tried the 8-E header and soldered pin 4 to 18 as recommended for higher quality (although i did also try the convenience setup but no noticeable difference.

ivan-gerov, 2022-12-17:
Hi guys, are there any updates on how to fix this?

hzeller, 2022-12-17:
Did you wire up the E line @ivan-gerov ?

ivan-gerov, 2022-12-17:
I have the Adafruit bonnet and I can't understand from the instructions how to solder the E line (I am sorry, I have limited electronics knowledge). So I can see I have E 24. I assume that is one half of the connection I have to make. To what do I have to solder E 24?

tidyhf, 2025-01-25:
I'm experiencing the exact same issue as you, with demo 10 emitting a small amount of light out of the unit sections.

Really disheartening to get this far and for it not to work, does anyone have any suggestions or maybe another HAT to buy?