hzeller/rpi-rgb-led-matrix #865: Led Matrix 32x16 Display issue

andriravoa, 2019-08-04:
Dear all,
I've a issue by displaying "forward scrolling an image" on my led matrix 32x16 with a RGB Matrix Panel Drive Board For Raspberry Pi (provided by electrodragon) using pi3 . 
During the test a use the following command: udo ./demo --led-rows=16 --led-chain=1 -D 9 runtext16.ppm and below is the screen i got. 
Could you guys please give an advise to fix it?
[image]
[image]
And this is my setup:
[image]

Thank you
A

hzeller, 2019-08-04:
This looks like a 1:4 multiplexed LED panel. Use the flag for led multiplexing and find the option working for your panel.

andriravoa, 2019-08-04:
Dear, after checking and using flag for led multiplexing. I found it and works with my panel. 
At the end, this is what i did: sudo ./demo --led-rows=16 --led-cols=32 --led-multiplexing=1 -D 3 runtext16.ppm
Thank you for your help. I appreciate your advise. 
Regards
A

closed 2019-08-04 by hzeller