hzeller/rpi-rgb-led-matrix #866: text scrolling on p3 32x64 doesn't fill  the high

rijotech, 2019-08-04:
Hi Mr Zeller 
I have a 32x64 led matrix and even after resizing and converting the fonts using otf2bdf,  i can't seen to get it to display corectly I was wondering how can I fix this since i have tried lots if different methods and nothig seems to work. 
sudo runtext.py --led-gpio-mapping=adafruit-hat --led-rows=64 --led-cols=32 --led-chain=2 -t "THIS MESSAGE!"

[https://ibb.co/V9FgJkH](url)

hzeller, 2019-08-04:
so if the font is too small, did you try to create it larger with otf2bdf ?

rijotech, 2019-08-05:
Yes I did increased the fonts size but instead of displaying from the bottom it does display from the same position as if it was a regular font. There is any switch to center the scroll text?

hzeller, 2019-08-05:
Just change the y-position when drawing the font. This is an example code for you to get started with; the example uses 10 as position.

closed 2019-08-05 by hzeller

rijotech, 2019-08-05:
Ok thanks Mr Zeller