hzeller/rpi-rgb-led-matrix #867: Not able do display anything on LED modules after RPI3 replacement with RPI4

uros-skopus, 2019-08-06:
Hi,

I am controlling my Chinese P5 outdoor module 192x64 (32x32, 6 cols, 2 rows) LED chain for months using RPI3 B+. I was using plain lite distribution of Raspbian Stretch mainly with Chinese HUB75-6 V1.2 (but also some with Electrodragon HUB - https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/ , both on picture below) and it worked quite decent.

[image]

Nevertheless, the only problem I had was the brightness issue, especially during sunny days, so I wanted to test new RPI4 to see if there was any improvement regarding this. 
I used plain lite Raspbian Buster OS, and tested firstly Flaschen Taschen server (since I was using it for months), and then just rgb-matrix part without success - no errors during server running, demos, led-image-viewer and other test utilities, just nothing is displayed...
For example after running:
`sudo ./led-image-viewer --led-rows=32 --led-cols=32 --led-chain=6 --led-parallel=2 --led-multiplexing=2 --led-slowdown-gpio=2 /home/pi/image.jpg`
I've got everything OK, but nothing is displayed:
```
Size: 192x64. Hardware gpio mapping: regular
Loading 1 files...
Loading took 0.066s; now: Display.
```

According to the RPI4 spec - the GPIO pins **retain their standard functions set by previous models along with extra functions for existing pins** (source - https://maker.pro/raspberry-pi/tutorial/raspberry-pi-4-gpio-pinout). Electrodragon hub is specified with: Support Raspberry Pi 2 and 3, most pins used for matrix driving, so still no RPI4 mentioned...

Just to mention I have disabled sound on RPI4 both under /boot/config.txt (setting dtparam=audio=off) and under /etc/modprobe.d blacklisting ALSA within alsa-blacklist.conf file.

Do you have any idea what else should I try or what can cause this issue?

hzeller, 2019-08-06:
did you get the latest version of this library when switching to the pi4?

uros-skopus, 2019-08-06:
2 days ago I've cloned the latest version of Flaschen Taschen server (with --recursive flag) used for my testing.
But now, I've removed it, and cloned latest version of just rgb-matrix and it works! 
Now I see that version 067eb95 committed on May 3, 2018 was attached to FT server, and that was the problem.

uros-skopus, 2019-08-06:
@hzeller thanks for a good hint! Will close this one.

closed 2019-08-06 by uros-skopus

Rodriguezjp, 2022-03-21:
He comprado una de estas placas pero no consigo hacerla funcionar con mi raspberry pi zero w, sabes si existe alguna web o manual que seguir paso a paso. mi pnel es de 320x160 .... se lo agradecia mucho

Rodriguezjp, 2022-03-21:
I bought one of these boards but I can't get it to work with my raspberry pi zero w, do you know if there is any website or manual to follow step by step. my panel is 320x160 .... I would appreciate it very much

mznmaurice, 2022-07-22:
@uros-skopus where can i buy this Chinese HUB75-6 part? Kindly share a link, thanks.