hzeller/rpi-rgb-led-matrix #868: Python bindings suddenly quit working

mansj, 2019-08-06:
I don't know why, but all of a sudden all Python bindings on my install quit working. The demo projects work as before, but any attempt to use the Python bindings do nothing - the LED is black.

I've reinstalled the bindings but no change. 

What could cause the Python bindings to quit working as long as the api-example demos work as they should, indicating nothing is wrong with the hardware or wiring?

MikeHLee, 2019-08-30:
I've got the same problem. API example demos work. We've followed all the instructions and nothing happens when we run sudo python image-viewer.py myimage.jpg --led-gpio-mapping=adafruit-hat --led-rows=32 --led-cols=64 --led-slowdown-gpio=2. The LED is black

vaseeece, 2019-09-08:
> I've got the same problem. API example demos work. We've followed all the instructions and nothing happens when we run sudo python image-viewer.py myimage.jpg --led-gpio-mapping=adafruit-hat --led-rows=32 --led-cols=64 --led-slowdown-gpio=2. The LED is black

I used the same command but hardware mapping is regular the image is not getting displayed properly there is more flickering but rest of the python examples work better..... @hzeller can you please help me out with it......
And the c++ program in utils work better I don't why the python example for image display is not working good.. please help me out.....