hzeller/rpi-rgb-led-matrix #869: led matrix 30x30,  scan mode 1/3

andrei0807, 2019-08-10:
I have led matrix of 30x30, scan mode 1/3.
I am going to control this on Pi using this library.
But it was not work correctly.
How to set params for this led matrix.
Please help me.
Thanks.

hzeller, 2019-08-10:
that is an unusual scan mode that I have not come across before; probably needs slight adjustments in the code. Can you take a picture of the connector on the matrix PCB including the labeling ? Do you have a datasheet ?

1/3 implies that we have essentially have in each half display 3 stripes of 5 led rows if it otherwise behaves as usual panels. Probably sufficient to just write another multiplex mapper.

andrei0807, 2019-08-10:
Thanks for your response,
I will send you datasheet now

andrei0807, 2019-08-10:
[image]
 Does it make sense?
Is this enough?

hzeller, 2019-08-10:
Yeah, this looks doable if it otherwise works with the library. If you connect it to the library, do you get any output (albeit garbled) when you run the demo ?

You essentially need to add a new multiplex mapper in `lib/multiplex-mappers.cc`. I am currently travelling (eventually hitting Chaos Communication Camp), so can't really help much right now.

andrei0807, 2019-08-10:
Thanks for your response and sorry to disturb your rest.
[image]
Should I add class like this?

andrei0807, 2019-08-12:
@hzeller Can you explain about parameter's mean and function's login?

hzeller, 2019-08-12:
See the header files for the corresponding class you implement.
A Mjultiplex mapper is based on a PixelMapper who's parameters etc are described in the corresponding header in detal
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/pixel-mapper.h

The Multiplex mapper is derived from that, and only adds one more method the `EditColsRows()`
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers-internal.h

.. there is already an implementation in `MultiplexMapperBase`, that implements that so you could instantiate that with 3 as parameter; Then the only function you need to implement is `MapSinglePanel()`, which takes the original x/y [0..29, 0..29] range and maps it to the new [0..149, 0..9] range. Something like this would be probably a good starting point:

```c++
class ThirdMultiplexMapper : public MultiplexMapperBase {
public:
  ThirdMultiplexMapper() : MultiplexMapperBase("Third", 3) {}

  void MapSinglePanel(...) const {
    // ... implementation according to your data-sheet
 }
};
```

Then register it in `CreateMultiplexMapperList()` and it should be available via the command-line flags.

angelogoncalve, 2019-08-12:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers-internal.cc

/*

 * Here is where the registration happens.

 * If you add an instance of the mapper here, it will automatically be

 * made available in the --led-multiplexing commandline option.

 */

static MuxMapperList *CreateMultiplexMapperList() {

  MuxMapperList *result = new MuxMapperList();

  // Here, register all multiplex mappers from above.

  result->push_back(new StripeMultiplexMapper());

  result->push_back(new CheckeredMultiplexMapper());

  result->push_back(new SpiralMultiplexMapper());

  result->push_back(new ZStripeMultiplexMapper("ZStripe", 0, 8));

  result->push_back(new ZStripeMultiplexMapper("ZnMirrorZStripe", 4, 4));

  result->push_back(new CoremanMapper());

  result->push_back(new Kaler2ScanMapper());

  result->push_back(new ZStripeMultiplexMapper("ZStripeUneven", 8, 0));

  result->push_back(new P10MapperZ());

  result->push_back(new ThirdMultiplexMapper("Third", 3));

  return result;

}

const MuxMapperList &GetRegisteredMultiplexMappers() {

  static const MuxMapperList *all_mappers = CreateMultiplexMapperList();

  return *all_mappers;

}

andrei0807, 2019-08-12:
@angelogoncalve 
Thanks for your response.
Can you write MapSinglePanel function for my led matrix?

andrei0807, 2019-08-12:
result->push_back(new ThirdMultiplexMapper("Third", 3));
Value 3 is correct? I think it's 5

andrei0807, 2019-08-12:
@hzeller 
How are you?
I added new MultiplexMapperBase  class but my LED doesn't work now.
Do I only have to add a new MultiplexMapperBase  class?
Don't I need any more settings?

hzeller, 2019-08-12:
How is your implementation @andrei0807 ?

andrei0807, 2019-08-13:
[image]
# sudo ./demo D 1 --led-gpio-mapping adafruit-hat --led-multiplexing 10 --led-rows 30 --led-cols 30
I changed code and used like command line.

andrei0807, 2019-08-15:
@hzeller 
Do you have time for me?

hzeller, 2019-08-31:
Did you get it to work ? If so, send a pull request, so that others can make use of it as well.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828