hzeller/rpi-rgb-led-matrix #870: Some patterns in LED display.

KeshavaShreenidhi, 2019-08-12:
I have attached a video clip in that we can clearly see some patterns are rolling when I display a image in the screen. Is there any solution for this?

[image]

hzeller, 2019-08-12:
how did you connect the panel? This looks like a typical case of using the Adafruit Hat without doing the pwm modification.

Also make sure to do the usual methods to minimize flicker, such as the isolcpus setting.

angelogoncalve, 2019-08-12:
Also beside hardware tricks to minimize the flicker  Adafruit Hat with doing the pwm modification:

--led-hardware-mapping='adafruit-hat-pwm'

do the change of theses flags:

--led-slowdown-gpio=5
--led-pwm-bits=1
--led-pwm-dither-bits=2

KeshavaShreenidhi, 2019-08-13:
Sorry I should have mentioned these earlier. I used Raspberry pi 3 B+ and seven P3 64x64 pixel led panels connected in a row. And I used the adapter board that you have mentioned ( https://raw.githubusercontent.com/hzeller/rpi-rgb-led-matrix/master/img/three-parallel-panels-soic.jpg ).
Also I tried isolcpus=3, but there is no much noticeable changes observed. 
I observed those patterns when I used this command : "sudo python  example_led.py --led-slowdown-gpio=2 --led-rows=64 --led-cols=64 --led-chain=7 --led-brightness=80 --led-pixel-mapper=Rotate:180 --led-pwm-bits=8 --led-pwm-lsb-nanoseconds=80"

I tried  --led-slowdown-gpio=5 but it shows "error : argument --led-slowdown-gpio: invalid choice: 5 (choose from 0, 1, 2)"
Also " error: unrecognized arguments: --led-pwm-dither-bits=2 " 

Thank you.

hzeller, 2019-08-13:
Are you doing anything on the Python code that is calculation intensive ? Python is very slow and taxing on the performance on other things. From the picture above it looks like you only show a static image though. If filled once and then you essentially sleep, that is ok (in general, using c++ is of course preferred to using Python. Only use Python if absolutely needed for something). 

If you do change image content it is helpful to use `SwapOnVsync()` so that you don't introduce tearing.

If just showing static content you can of course simply use the provided led-image-viewer.

64x64 panels in a row is lot as these panels do 1:32 multiplexing, so a lot has to be clocked in per row. Using parallel chains and some pixel mapping can help.

Slowdown gpio will not bring advantage in the artifacts you see.

I suppose you did all the things mentioned in the readme in the troubleshooting section: not using gui, using minimal os, switching off one-wire etc. ?

angelogoncalve, 2019-08-13:
For minimal OS used DietPi is not intensive for the disk of RPi3B+

KeshavaShreenidhi, 2019-08-14:
I am using python but there is not much calculations involved. Even I tried the c++ version but no luck. 
Previously I was using vnc viewer but now i am not using it and i did almost all the things in the troubleshooting section and problem not resolved.

@angelogoncalve I'll definitely try and let you know.

Thank you.

KeshavaShreenidhi, 2019-08-15:
Whether putting an raspberry pi 4B (4GB ram) instead of rpi 3 b+ will solve the problem?
Does this library supports raspberry pi 4B?

angelogoncalve, 2019-08-15:
RPi4B has more CPU.

Yes this driver supports Rpi4B.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin