hzeller/rpi-rgb-led-matrix #871: How to debug this library ?

GeorgeFlorian, 2019-08-14:
Hello !

I am trying to port a working project form Raspberry Pi Zero to a Banana Pi BPI M3.

I was able to "activate" the desired pins on the Banana Pi for a single RGB panel. But I still have 0 output on the Panel and also 0 signal on the pins. So the project is not pushing a thing to output.

How can I debug this ? Is there any debug tool ? Can I see what's different from the Raspberry to the Banana on the library side ? What or how does it work ?

hzeller, 2019-08-15:
You will not be able to make it work without modifying gpio.cc and understand in detail how the SoC of the Banana Pi works.

The Banana Pi is totally incompatible with the Raspberry Pi, they only happen to have also 40 pins as GPIO pins to look similar to the Raspberry Pi for marketing reasons.

So In order to run it with a different computer, you need to read the datasheet of the SoC of the Banana Pi and implement gpio.cc for that platform. You need to find if it is possible to memory map the GPIO output. Also, you need to see if the Banana Pi has a programmable timer that you can use and can be pinmuxed to an GPIO pin to be able to generate the output enable pulses.

You need to be versed in reading SOC datasheets and know low-level programming for this to work.

GeorgeFlorian, 2019-08-15:
Great answer ! Thank you for breaking down the things that are needed in order to make the Banana Pi function with the library.

It is a lot more complicated than I thought and it is way beyond my current skills.

I will stick to the Raspberry Pi 4.

closed 2019-08-15 by GeorgeFlorian