hzeller/rpi-rgb-led-matrix #872: PWM issue

Sai5157, 2019-08-17:
Hello hzeller,
I have been using the above code to generate an image on 6 panels(2 rows*3 panels) using U-Mapper and I do run some background code in python along with yours. But when I see the image is fine when a single image is played for a longer duration. But when I play an image for a shorter duration or play  2 or more images in a slide format. I see a small flash kind of thing(Image itself flashes) during the transition. Is it because of the pulse width modulation and if Yes may I know if lowering the number using --led-pwm-bits=<1..11> will solve the problem?

hzeller, 2019-08-17:
How do you do the transitions ?
Usually the best is to fill in the new image in a frame-canvas, then do `SwapOnVSync()`

Sai5157, 2019-08-17:
Generally, each image is set for 2 mins to play, even when a single image playing for 2 mins I see the flash happening frequently.

hzeller, 2019-08-17:
How do you do the transition ? Do you stop and restart the matrix (because of course, then you'll see flashes), or are you pre-filling the next image in a FrameCanvas, then do the `SwapOnVSync()` ?

Do you use the [active3 adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) to connect the panels ?

Be aware that Python uses more resources, so you have to use all the tricks to make sure to not influence your pictures by pre-calculate frame canvases etc.

If you just want to show images with a time in-between, consider just using the provided [`led-image-viewer`](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#image-viewer).

(In general, consider writing your program in C++. It is easier than Python and gives you more performance advantages.)

hzeller, 2019-08-17:
(also, I suspect you already do all the things suggested in the README, such as no GUI, switching off one-wire, using isolcpus, using e.g. dietPi, etc...)

Sai5157, 2019-08-17:
I am unable to provide the video now. I will provide it as soon as possible. May I know one thing why is --led-pwm-bits= is restricted between <1..11> what happens if we increase it to higher value.?

hzeller, 2019-08-17:
refresh would be too slow.

closed 2019-08-17 by Sai5157

angelogoncalve, 2019-08-18:
Use threads to do the transition next image from the video:

https://eli.thegreenplace.net/2011/12/27/python-threads-communication-and-stopping