hzeller/rpi-rgb-led-matrix #878: Not working properly with 64x32 p4 display

vasee007, 2019-08-29:
I'm using a p4 64x32 display, when when I connected initially it went right but after rebooting the RPI it's not working...... When I run a demo it shows timeout reached.... Plz help me......☹️😥....

hzeller, 2019-08-29:
What do you mean with 'timeout reached' ? The matrix program should never show this message.

Did you do all the things mentioned in the troubleshooting section, i.e. switching off unnecessary IO ports such as 1-wire ?

vasee007, 2019-08-30:
I have disabled onboard audio but I didn't see any 1-wire thing in RPI config.......
maybe I think that I should change the minimal version of os as mentioned and try again......
please keep in touch and help me with these things as I'm doing a project based on this for my college...

Thank you!!!!....

vasee007, 2019-09-04:
Hello, I'm getting that timeout reached error again after ensuring all the troubleshooting measures.....
i've attached a screenshot below plz have a look at it and reply me with a
[image]
 solution asap..... plz help me...

hzeller, 2019-09-04:
Uhm, you explicitly *told it* to run the demo for only 20 seconds with the `-t 20` option. So it ran for 20 seconds, then timed out. Working as expected :)

vasee007, 2019-09-04:
No, actually I can send you the video
actually
it's not working...

Sorry if you don't mind can I get your WhatsApp contact so that i will in
abetter position to explain my issue...
please help me sir you are my only hope to complete this project and my
college fullfillment.... please dont hesitate sir im really trying out very
hardly....

Beersatron, 2019-09-04:
I think hzeller is saying that you should remove "-t 20" from your command and then run it.

hzeller, 2019-09-04:
If it is not working otherwise, please read the docs and first go through the troubleshooting section and address all the potential issues mentioned there.
Look closely what type of panel you have to set the parameters correctly.
How did you connect the panel ? You must've either soldered up an [active3](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) adapter or wired up an level shifter by yourself when connecting. Double-check first that these signals are connected as intended.

Also, I see that you are using a desktop system, which is also known to break things. Install a simple Rasbpian or DietPi first and only interact with it using the ssh.

If you file a bug always make sure to include all relevant information.

vasee007, 2019-09-04:
I'm using a P4 32x64 Panel and i connected it via means of jumper cables as per wiring guide given
and initially the volume bar and rotating square examples worked well and I shut down pi for some reasons... later I connected it and then started the issue...

hzeller, 2019-09-04:
Cool, so it is working in general, but be aware that connecting it directly to the pi also means that you only have 3.3V logic level (which might or might not work reliably depending on the panel), and that you essentially directly wire up the long cable load to the output drivers of the processor - if there is any glitch or too strong reflection, it is easy to break its output drivers. So I strongly suggest to always put it through a level converter. If there are 5V pull-ups on the LED matrix side, they can also interact badly with the Pi. So if you suspect this might be the issue, tests with a different pi.

Make sure to boot it into _non_-desktop mode and disable all the relevant interfaces (such as 1-wire) using `raspi-config`.

vasee007, 2019-09-04:
> Uhm, you explicitly _told it_ to run the demo for only 20 seconds with the `-t 20` option. So it ran for 20 seconds, then timed out. Working as expected :)

The timeout reached came for that only... sorry I checked now but I had no output on the panel so I thought it was the issue and i ran other demo and i saw no timeout but there is no proper output on the screen......

closed 2019-09-04 by vasee007

vasee007, 2019-09-05:
> Cool, so it is working in general, but be aware that connecting it directly to the pi also means that you only have 3.3V logic level (which might or might not work reliably depending on the panel), [...]

Do you mean that it could break or damage the raspberry pi??

hzeller, 2019-09-05:
If you accidentally feed in 5V into the GPIO pins it easily can, yes.

vasee007, 2019-09-05:
Well in our case the raspberry pi is only feeding the led panel right ?? and we are not giving any input to raspberry from panel.... or if im wrong can just please explain it without hesitation??.....

hzeller, 2019-09-05:
if there are pull-ups on the input of the panel, they will be on 5v level. Not saying that this is happening, but it is something to keep in mind.

I'd be more worried about large capacitance to drive long cables from the SoC 

Anyway. your problem might be much simpler e.g due to cabeling.

vasee007, 2019-09-06:
Just for the clarification can I use this level shifter (image has been
attached) for rest purpose or should I use only 74ahtc245 level
shifter??....

vasee007, 2019-09-06:
> if there are pull-ups on the input of the panel, they will be on 5v level. Not saying that this is happening, but it is something to keep in mind.
> 
> I'd be more worried about large capacitance to drive long cables from the SoC
> 
> Anyway. your problem might be much simpler e.g due to cabeling.

Just for the clarification can I use this level shifter (image has been
attached) for rest purpose or should I use only 74ahtc245 level
shifter??....
[image]

hzeller, 2019-09-06:
You don't show the specifications of the level shifter, so it is not possible to judge from the picture. But you want something that can convert 3.3V logic level to 5V logic level and is at least 30Mhz fast.

vasee007, 2019-09-06:
Ok then here is the link of that level shifter it's from sparkfun....

https://www.sparkfun.com/products/12009?_ga=2.26660287.367488503.1567749182-930865824.1566889831